opentrack/opentrack #1141: Easy Tracker Vertices

whiskeygreg, 2020-08-28:
Hello,

Very cool program. The basic Kinect Facetrack is awesome. I was so shocked at how much of a game changer head tracking is in DCS World. After looking into it a bit more, I decided to use Easy Tracker. I can't seem to get it working properly, though. I believe my main issue is being confused about the settings. First, when I click on the settings under Input: Easy Tracker 1.0, they are labeled as PointTracker Settings. Second, the model tab. I have three vertices selected, but I am confused about the measurements. 

Using the ones I've found in these forums before has not yielded good results, despite me using a hat very similar to the picture floating around here (A small ball wrapped in reflective tape on the top middle of the hat, and two more on the bill, one on each side, looks like Splinter Cell haha). Are you able to explain it a bit better? I am able to figure out the mapping, for the most part, but there is also a bit of roll input when I give yaw, and the X axis moves me in and out instead of side to side.

Any help would be appreciated. This is a sweet program and has been fun to tinker with over the last week, but I admit that I may need some help!

Cheers

Slion, 2020-08-28:
Can you share your settings, vertices, hat, solver? Make sure the vertices are matching your hat tracking points dimensions. Make sure you are far enough from you Kinect. I think it's at least 70cm but maybe 100cm is better.

whiskeygreg, 2020-08-29:
I'm using P3P, 

[image]
[image]
[image]

Slion, 2020-08-29:
Maybe around 70cm from sensor is best.
I just tested an older dev build on my machine and it was working great with the following settings.
Could be the size of your vertices are causing problems somehow.

[image]

Slion, 2020-08-29:
Are you sure your model definition is actually matching the position of your trackers on your hat? Looking at your screenshot it looks like they are closer together than on mine and you have the exact same vertices definition which looks rather suspicious.

You need to measure the distances between the centre of each trackers.

[image]
[image]

whiskeygreg, 2020-08-29:
I have adjusted my hat to be more in line with yours. My vertices are definitely bigger, but they are now spaced further apart. This did help, however my side to side movement is still interpreted as forward and backward.

Slion, 2020-08-29:
> my side to side movement is still interpreted as forward and backward.
> a bit of roll input when I give yaw, and the X axis moves me in and out instead of side to side.

I see what you mean now. I do believe I have the same issues. No idea why, possibly a failure in OpenCV solver. The most obvious one is indeed Z acting up if you are off to the left or to the right.

I really ought to try and deploy other solvers:
https://github.com/opentrack/opentrack/issues/1020#issuecomment-573602447

whiskeygreg, 2020-08-29:
I'll wait with bated breath. Thanks for your help! This is such a cool project, even if my wife says I look silly with the hat on.

Cheers!

Slion, 2020-08-29:
> No idea why, possibly a failure in OpenCV solver or our inputs.

Though to be honest it is hard to think OpenCV solver would be that bad so I'm pretty sure we are doing something wrong.

whiskeygreg, 2020-08-29:
If you end up figuring it out, please let me know. I think I'm almost getting the hang of this, but it's all a bit above my level.

Cheers

Slion, 2020-08-29:
I just tried with 5 markers and the UPNP solver and that's the same issue. Stride left and right screws our Z position.

whiskeygreg, 2020-08-29:
I really appreciate you taking your time to mess around with it.

Slion, 2020-08-29:
I found the problem, it has to do with the way we feed the distortion coefficients to OpenCV solver.
Just disabling distortion coefficient usage makes it work great. Ideally we should fix them but disabling is fair enough for now. You should only get noticeable problems when moving to the far edges of the frame which is hardly a use case for us.

Slion, 2020-08-29:
> I really appreciate you taking your time to mess around with it.

Thanks for taking the time to report that issue in the first place and challenge us to come up with a solution.

whiskeygreg, 2020-08-29:
> Moreover it could take me some time to contribute this back to the main branch has I've not been keeping with the project for a while. I just patched it on an old locale checkout I had on my machine.

I am always happy to wait for improvements. Is there a workaround that I could do in the meantime? Otherwise I am happy to go back to plain face tracking for now.

Thanks again! You've been so helpful!

Slion, 2020-08-29:
You should be able to download it from there:
https://slions.net/threads/open-track-for-kinect-v2.67/

whiskeygreg, 2020-08-29:
Thank you! I don't think I'm relying on anything, I'm just winging all of it. Should I uninstall what I've got and reinstall this one? Or is this a matter of dropping files in folders?

Cheers

Slion, 2020-08-29:
I'm actually shocked at how well it works now, that thing always bugged me but could never quite put my finger on the issue. I'm off to try it in MWO.

For reference the fix involve disabling the following lines in function `Tracker::CreateCameraIntrinsicsMatrices`:

    //for (unsigned k = 0; k < 8; k++)
    //    iDistCoeffsMatrix(k) = (double)iCameraInfo.dist_c[k];

Slion, 2020-08-29:
> Should I uninstall what I've got and reinstall this one? Or is this a matter of dropping files in folders?

Just unzip it somewhere and run it from there, it's not an installer, just the binaries. Copying the files over your current installation may or may not work so I don't think you should bother. Just run it from wherever you want until the fix makes it into an official release.

whiskeygreg, 2020-08-29:
Copy that, thank you!

Slion, 2020-08-29:
Hope the download works though. I never published files that large on my web site before.

whiskeygreg, 2020-08-29:
I am getting these errors when trying to run from your binaries.

[image]
[image]
[image]
[image]

whiskeygreg, 2020-08-29:
I can run the latest version. Perhaps I'd be able to copy a specific file over to enact the change you made?

Slion, 2020-08-29:
I've added another zip file with the missing DLLs just unpack them next the the EXE.

whiskeygreg, 2020-08-29:
Awesome!

New problem (naturally), it opens now but Easy Tracker is not listed! I can't win.

Slion, 2020-08-29:
I guess some other dependencies are missing somehow.

whiskeygreg, 2020-08-29:
Well, I've been messing around with it for the last hour or so and cannot seem to finagle a way to make it work, unfortunately.

helmick43270, 2020-08-30:
Hello, Slion and whiskeygreg,

This is amazing to read a conversation between people trying to figure out the nitty-gritty details to make something work that, for me, has been cerebral reality for around 24 years. This is something I knew could be done, and could visualize the user experience in my imagination back in 1996. I haven't kept up with developments on all fronts very well, so I watched a YouTube video of Kinect face track on a flight simulator. It seems that the effect is still fairly short of what it could be if there were different images presented to each eye, using glasses that alternately flicker. Are frame rates high enough yet to prevent headaches in that configuration? Anyway, I apologize that I can't contribute any technical details to help, but I just wanted to say that it's encouraging to me to see details being hammered out in the formation of the state of the art, and of today's inspired reality.

whiskeygreg, 2020-08-30:
Helmick, I'm not even almost qualified to answer your question, but thank you for the kind words!

Cheers

Slion, 2020-08-30:
@whiskeygreg Try installing various [Visual Studio runtimes](https://www.microsoft.com/en-us/download/details.aspx?id=30679) until it works.

whiskeygreg, 2020-08-30:
No joy :(

Slion, 2020-08-30:
> No joy :(

You tried all those redistributable and Easy Tracker is still not loading?

whiskeygreg, 2020-08-30:
Correct. I even went through and uninstalled them all to reinstall them fresh, one at a time.

Slion, 2020-08-30:
> I apologize for taking up so much of your time!

I'm sorry you could not get it to work on your machine. I'm giving up trying to solve that deployment issue for now. I've pushed that fix so hopefully you will get a new release soon. Maybe @sthalik can help us with that.

whiskeygreg, 2020-08-30:
You have done more than enough for me! Something to note, while I still had issues with Z and X axis movement, adjusting the curves to a pretty significant level did help some. Still, it was tough to fly and track people in DCS when I started moving more freely. 

I appreciate you pushing the fix. Cheers to @sthalik as well.

Slion, 2020-08-30:
@helmick43270 
> It seems that the effect is still fairly short of what it could be if there were different images presented to each eye, using glasses that alternately flicker. Are frame rates high enough yet to prevent headaches in that configuration?

Well that's what active 3D TV, monitors and projectors are doing. So yes frame rates are high enough. However this days people usually go for HMD and virtual reality instead.

helmick43270, 2020-08-30:
Ok, wow, thanks very much! @Slion

Slion, 2020-09-01:
@whiskeygreg I've published a brand new ZIP on that same web page. Try it out and see if it works this time.

whiskeygreg, 2020-09-01:
Thanks, @Slion. I'll check that out when I get home from work today! You rock!

whiskeygreg, 2020-09-02:
@Slion It seems to not be tracking at all, unless I'm missing a step. It's all there, and I can start it, but the octopus won't move. I'll keep troubleshooting.

whiskeygreg, 2020-09-02:
@Slion Alright, after resetting my curves, it seems to be tracking now. I'll adjust from there and let you know. Thanks again! This is very cool!

Cheers!

whiskeygreg, 2020-09-02:
I do have one more question @Slion : Can you define the vertices a bit better? On the model page, under the X Y Z where we have the Top, Right, Left rows. How exactly am I to interpret that? I gather that it is a measurement for my hat, but I am not fully understanding.

Thanks!

Slion, 2020-09-02:
Each marker is matched to a vertex in the model the camera is tracking. To be able to compute the model pose and thus the pose of the head it is attached to, Easy Tracker needs to understand the model geometry.

Therefore you need to tell Easy Tracker the arrangement of your markers. This is done by giving your markers coordinates in the proper coordinates system of arbitrary origin. I recommend you use the top marker as your origin and therefore set it's coordinates to (x:0,y:0,z:0), then work out the coordinates of the two other markers. Units are in millimetres.

Think of it like this:
* X axis is horizontal and parallel to your Kinect, left is negative and right is positive.
* Y axis is vertical and parallel to your Kinect, up is negative down and down is positive.
* Z axis is horizontal and perpendicular to your Kinect, forward is negative and backward is positive.

The hat picture I posted above was supposed to illustrate that and matching the vertices coordinates you posted further up but confusingly I swapped the 10cm and 9cm labels  :)
Hope it helps.

PR #1143 (2020-09-02): tracker/easy: Disable Kinect distortion coefficients usage
Thus notably fixing badly broken Z component in translation vector.

Close #1141
Close #1020
commit e25f8189 (2020-09-02): tracker/easy: Disable Kinect distortion coefficients usage thus notably fixing badly broken Z component in translation vector.
Close #1141
Close #1020
closed 2020-09-02 by Slion