opentrack/opentrack #1145: Easy Tracker - Add clip support

NoX321, 2020-08-31:
Hi,

I'm trying to get a Kinect V2 working with OpenTrack. 
I made a clip which is perfectly detected in easytracker, but i'm unable to get the correct configuration.

Referring to the Easy Tracker wiki page, clip don't seems to be supported (unless someone "feels free to ask").

Signal seems clear enought to be used as input, but neither I wasn't smart enough to find the good setting nor it's simply not designed to work.

P.S. I wasn't able to find any mocap markers anywhere on the web. Any clues to where find some ?

Thanks for your help

Slion, 2020-08-31:
Show me what your clip looks like. I believe easy tracker three vertices model is expecting them aranged as a triangle.

Slion, 2020-08-31:
> wasn't able to find any mocap markers anywhere on the web. Any clues to where find some ?

https://optitrack.com/products/motion-capture-markers/
https://www.qualisys.com/hardware/accessories/passive-markers/

Looks like most people are building their owns using small balls, reflective tape and velcro stickers.

NoX321, 2020-08-31:
Here is what it look like : https://www.thingiverse.com/thing:1431730 
Slightly tweaked in my case, but shown dimensions are correct.

Slion, 2020-09-01:
> Here is what it look like

Can you show me what it looks like in the easy tracker preview? I guess I should build myself one of those and try it out.
If I recall well it would need a new algorithm for matching the vertices. The current one assumes arangement for hats, we need one that would work for clips.

NoX321, 2020-09-01:
What do yo need ? Video or just a picture ?

If you're in France (or in Europe), I can send you the printed part if needed.

Slion, 2020-09-01:
> What do yo need ? Video or just a picture ?

Screenshot would do.

> If you're in France (or in Europe), I can send you the printed part if needed

Thanks, I guess I could first try without it.

Slion, 2020-09-02:
I've implemented clip support and published a demo version [there](https://slions.net/threads/open-track-for-kinect-v2.67/). Make sure you [define your vertices properly](https://github.com/opentrack/opentrack/issues/1141#issuecomment-685762072).

[image]

NoX321, 2020-09-02:
Woaw, that was quick. I even hadn't time to send you what the clip look like. Here I go ...
[image]

Just downloaded and tried to launch : everything seems OK. Going to test it tommorrow and to make any feedback needed !

NoX321, 2020-09-05:
Hi,

Just took some time to play with this build. 
Sadly, the tracker is pretty unusable.

I took a live capture of the behavior : the spots seems to jump to one led to another randomly : 
https://drive.google.com/file/d/17mBYC-UwPOOJFRe3nlcML1ma3JSNn7sq/view?usp=sharing

You'll also find my settings below : 
[image]
[image]

I took measurments and i think i'm OK with clip dimensions.

closed 2020-09-06 by Slion

Slion, 2020-09-06:
Could it be that you swapped X and Y axis?

Slion, 2020-09-06:
I just gave it another try myself using the model below and it looks indeed like the solver does not like those kind of models. Most likely because it is just a two dimensions model. It is not as stable as the hat model. However after trying harder, adjusting models dimensions and adding some tracker deadzone it feels usable. Roll is being funny though but I reckon that could be compensated to satisfactory levels using mappings.   
[image]

commit 4f4c21ab (2020-09-06): tracker/easy: Adding support for Clip models where three vertices are arranged from top to bottom.
Close #1145
PR #1148 (2020-09-06): tracker/easy: Adding support for Clip models
Adding support for Clip models where three vertices are arranged from top to bottom.
Close #1145
commit abb14f0b (2020-09-06): tracker/easy: Adding support for Clip models where three vertices are arranged from top to bottom.
Close #1145