opentrack/opentrack #1146: TrackIR compatibility Ubuntu 18.04

mauricioduarte01, 2020-08-31:
Hello,
First I would like to thank the dev and all those involved in this project for the great work!
I have succesfully compiled and installed Opentrack on my machine but I don't see any option to make TrackIR to work on Linux. 
Can someone please help me with this? I compiled latest build from Opentrack using Opencv 4.

Thank you.

Slion, 2020-09-05:
> TrackIR

By that you mean the TrackIR camera harware? I'm not sure sure there is any public driver for it so it is most probably not supported.

sthalik, 2020-09-05:
> I'm not sure sure there is any public driver for it so it is most probably not supported.

https://github.com/uglyDwarf/linuxtrack

Slion, 2020-09-05:
> > I'm not sure sure there is any public driver for it so it is most probably not supported.
> 
> https://github.com/uglyDwarf/linuxtrack

Is there a driver for TrackIR camera in there? Do you know if there is a Windows driver for it too? If a Windows driver exists for the TrackIR camera I would interrested to add support for it in Easy Tracker.

mauricioduarte01, 2020-09-05:
> > TrackIR
> 
> By that you mean the TrackIR camera harware? I'm not sure sure there is any public driver for it so it is most probably not supported.

The developer of Linuxtrack managed to make TrackIR compatible on Linux, it works very well specially with X-Plane but the integration with "Steamplay/Proton" is not there yet and I think to make that work it will be necessary to create a way to make it compatible. 
AFAIK is natively compatbile with Steamplay/Proton, but the issue is that my TrackIR is not being recognized.
Thanks for everyone for the answer

@Slion you can check the source code.

mauricioduarte01, 2020-09-21:
> > > I'm not sure sure there is any public driver for it so it is most probably not supported.
> > 
> > 
> > https://github.com/uglyDwarf/linuxtrack
> 
> Is there a driver for TrackIR camera in there? Do you know if there is a Windows driver for it too? If a Windows driver exists for the TrackIR camera I would be interrested to add support for it in Easy Tracker.

Hello @Slion  sorry for mentioning you but i would like to know if you managed to find what you were looking for regarding linuxtrack and trackIR. If you need any help just let me know. Thank you