opentrack/opentrack #1151: Opentrack has to be the active window

Nts1GitHub, 2020-09-04:
Hello, I use droidcam + aitrack + opentrack to play microsoft flight simulator.
After tweaking and disabling game mode, i got it to work quite good but the opentrack window has to be the active window.
So i have to run MFS in window and have the opetrack on screen as the active window, if I don't, I get a 1-2 sec lag od movement ingame.
I have set the cpu priority on opentrack and aitrack to high but that didn't help.
Any idea what could be the problem?
Thanks.

sthalik, 2020-09-05:
Make sure that "game mode" in Windows 10 is disabled.

Nts1GitHub, 2020-09-05:
Yeah "After tweaking and disabling game mode", the problem is still there.
But, yesterday i played again and the difference between the active window and non active window lag was not that big.
I don't know, maybe it's a MFS thing it being microsoft and all :)

sthalik, 2020-09-05:
How many cores does MSFS use?

Nts1GitHub, 2020-09-05:
All of them i think. I have a 7700k

sthalik, 2020-09-05:
What's aitrack though? I'm suspecting something here.

Nts1GitHub, 2020-09-05:
https://github.com/AIRLegend/aitrack

coldblade2000, 2022-12-24:
For posterity, I fixed this by turning off GPU scheduling.

On Windows 10: Display Settings -> Graphics Settings (found at the very bottom) -> Turn OFF Hardware-Accelerated GPU Scheduling