opentrack/opentrack #1152: Profiles Cease to Work

Edwired, 2020-09-04:
I'm new to open track, managed to set up a profile I'm somewhat happy with but still has a long way to go.

However when I stop tracking, to tweak CL Eye Test settings, and then resume the profile that was once working no longer gets picked up. 0 LED points tracked, no movement what-so-ever. 

Copy the .ini settings to a new profile doesn't work either. Only creating a new profile and transferring settings across. It's like it locks up at the thought of other profiles.

Any ideas and is this a known bug?

Cheers, Ed

sthalik, 2020-09-05:
Make sure you don't use Windows 10 OneDrive integration for `Documents` directory.

Edwired, 2020-09-05:
Nope my OpenTrack is stored properly on physical drives so there is no danger of that.