opentrack/opentrack #1153: Opentrack shows in task manager but not on screen.

RinoStoof, 2020-09-05:
Hi, first post here.
I downloaded Opentrack with A.I. Track to use in MSFS 2020. It worked well on my first try, great job!
This morning, when I start Opentrack nothing showed on the screen/task bar but it was in the task manager list.
Killed it, tried to delete the directory and got told some file was still open.. Which file?
Re-booted, deleted directory, re-installed (copied folders from compressed file), same behavior.
The update went in before I did this so there was no changes to the sim.
A.I. Track works fine..

So.. What have I done wrong?

Thanks very much for your help!

Rino
NZ

sthalik, 2020-09-05:
Send me the minidump. See https://www.wintellect.com/how-to-capture-a-minidump-let-me-count-the-ways/ for the Task Manager method.

RinoStoof, 2020-09-05:
Hi sthalik
I tried to send it b changing the extention to txt but it is 136MB. Even zipped it's still 52MB. How can I get that to you?

standaman94, 2020-09-06:
I'm also having this issue, installed and worked fine yesterday but today the window just isn't opening, it's there the in background processes though.

RinoStoof, 2020-09-10:
I went back to the previous version and that seems to be fine, thanks for your reply! :-)

closed 2020-09-10 by RinoStoof

diVineProportion, 2020-12-17:
@RinoStoof does the process in Task Manager stay there indefinitely or does it show then disappear?

@sthalik I commented elsewhere about my issue, but ill continue here. It seems that opentrack has no problem starting on a fresh reboot which leads me to think that some event in the future is causing the crash. Can see the process appear then disappear, but never the GUI. I got around this by starting opentrack elevated.

0xC0000409
[opentrack.exe.14824.dmp.zip](https://github.com/opentrack/opentrack/files/5706880/opentrack.exe.14824.dmp.zip)

or do I need to use the method mentioned above?

RinoStoof, 2020-12-18:
Hi

I'm afraid I haven't been using opentrack as I find all the movement to restless..

Thanks for looking into it though!

Cheers

Rino