opentrack/opentrack #1154: raw input flip the model during yaw

JackPlay09, 2020-09-05:
Hi
this is when centered
[image]
while moving yaw axis the model turn ok
[image]
at certain range of yaw the cross change attitude, like the head flipped upside down
[image]
yaw on the other side same result after reaching some range of turn
[image]
the model measures are as accurate as possible
[image]
tried calibrating a lot of times bus same result.
Re-done the model to position the high center led to get the same proportions of default cap measures 100,60,40, but can't go as much far as 73/75 mm from front, otherwise I have to trash the cap and start from new. I should have 90,54,36.
Any suggestion on my setup or is a bug in PointTracker?
Thanks
Jack

eifiionglyn, 2020-09-12:
I have this same issue, first started without any changes to the system while running 2.3.11, updated to 2.3.12 in the hope of a fix with no positive outcome.

Tracker finds points from the camera fine, and the video preview is smooth. Raw data for rotations increase and decrease smoothly as expected while turning my head, but at certain amount of yaw the output data jumps, so you'll get like 8 degrees input = 30 degrees output, then 9 degrees input will give you 60 output.

I have a slight expo curve to lower the response around centre, but the behaviour is also seen when using liner mapping.

Have uninstalled / reinstalled with no positive outcome, tried fresh profiles with no positive outcome.

Opening the point tracker input settings shows the listed camera FPS being very erratic, jumps from 75 to 25 to 3 and back up again. (should be solid 75 with PS Eye camera)

I don't think it's a camera issue because:
Tracking with same camera and clip is fine in FacetrackNoir.
Same erratic jumping in output data is seen when specifying the test tracker as an input. This puts a gradually varying input which should result in smoothly varying output data, instead I get jumping output data accompanied with a rapidly flipping track model. At input yaw values over 40 degrees I get the output flipping between -176 and +176. Test tracker should have no impact from my camera or point model.

What's odd is that this started happening spontaneously, was working perfectly previously.

whiskeygreg, 2020-09-27:
Have you tried using Dynamic pose under the camera tab?

DJDC13, 2021-04-19:
I think this is due to the tracked points overlapping each other at certain angles combined with the spread angle of the IR LED

thatsenough1, 2025-04-17:
have the same issue with reference 3-points cap on WiiPointTracker ver 2022.1.1, this is not caused by point overlapping, or Dynamic pose(WiiPointTracker plugin does not have that option). It seems the reference point(the one with yellow cross in preview) get changed from top point to the nearest point when yaw angle exceed some value, this only happens on 3-point cap, not vertical clip,
it should be easy to pinpoint change responsible if a previous version does not have this issue.