opentrack/opentrack #1155: Opentrack shows in task manager but not on screen

nathanLH458, 2020-09-05:
Launched well the first time. After closing it once, it won't start again.  
I have tried restarting, rebooting, etc...  
None works.  
  
Dump file: to big for upload here, so google drive: https://drive.google.com/file/d/1jIDzT0TgNECu0uKxSI457uTDL_WWTFvw/view?usp=sharing

Hope this helps, thanks.

nathanLH458, 2020-09-05:
To be clear because my post seems confusing after reading it again. It does technically start up, shows up in task manager, but now way for the window to pop up (so unuseable)

AF1E, 2020-09-09:
I have the same problem  I uninstalled Opentrack and reinstalled. The installation is showing two errors.  Like nathanLH458 I was able to see the screen on my first try but have not since.

First Error
Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x5f3e37fd
Faulting module name: opentrack-video-opencv.dll, version: 0.0.0.0, time stamp: 0x5f3e3849
Exception code: 0xc0000005
Fault offset: 0x00059c3e
Faulting process id: 0x2350
Faulting application start time: 0x01d68699b57da0d8
Faulting application path: C:\Program Files (x86)\opentrack\opentrack.exe
Faulting module path: C:\Program Files (x86)\opentrack\modules\opentrack-video-opencv.dll
Report Id: 65195de8-c103-48da-899f-1206ef6d3e83
Faulting package full name: 
Faulting package-relative application ID: 

Seccond Error
Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x5f3e37fd
Faulting module name: opentrack-tracker-wii.dll, version: 0.0.0.0, time stamp: 0x5f3e3852
Exception code: 0xc0000005
Fault offset: 0x00001025
Faulting process id: 0x5d84
Faulting application start time: 0x01d6869b40a197c2
Faulting application path: C:\Program Files (x86)\opentrack\opentrack.exe
Faulting module path: C:\Program Files (x86)\opentrack\modules\opentrack-tracker-wii.dll
Report Id: 9337a71e-e08b-482a-a03f-9e2dac2c34f0
Faulting package full name: 
Faulting package-relative application ID: 

Task manager shows the app is running, but I cannot see the config screen.

epaga, 2020-09-09:
Is it possible it's in the system tray?

Aaron-Davison, 2020-09-13:
Yeah I just downloaded and installed and am having the same issue. Doesn't open at all. Not in the taskbar or window. Just shows in Task Manager

Aaron-Davison, 2020-09-13:
So its open now. All I did was another restart and also added a Firewall rule to allow traffic to and from the app.

sthalik, 2020-09-28:
Can we confirm if disabling Windows Defender helps?

wftinmn, 2020-09-29:
I have the same problem. Tried uninstalling and reinstalling and the program has not worked since very first time. It does show up in the task manager. 
Turning off windows defender had no effect.

PhilTNNY, 2020-10-03:
Same issue here. Disabling Defender, cold boot doesn't help.

I was following @epaga instructions at inflightassistant.com. I downloaded the software and then the firewall instructions. Warm boot, went to open OpenTrack and it wouldn't load. I did see it in the system tray but then it disappeared. Went ahead with a cold boot, removed the firewall rule, then uninstalled. Reinstalled and check the box to open when complete, this time it won't open.

sthalik, 2020-10-04:
So do you have system tray support enabled in opentrack?

(sorry, I'm not able to work on the software in any serious capacity lately.)

PhilTNNY, 2020-10-04:
I can't get into OpenTrack. Initially it was showing in the system tray but as I opened the tray it then was removed.

diVineProportion, 2020-11-05:
I've on and off (cause I format like once or twice a year) had this problem and haven't been able to figure out why, but **starting as administrator** seems to be a temporary fix if indeed we are experiencing the same problem

BeetleBailey227, 2024-10-11:
The Opentrack icon/UI is in the systray  on the taskbar. Open the systray and drag the icon to the taskbar and it should be accessable from there