opentrack/opentrack #1156: Windows 10 Event Error

Len62, 2020-09-06:
Found this while looking through the event viewer. Appears to happen each time I start tracking. Looks like something to do with the Wii dll.

Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x5f3e37fd
Faulting module name: opentrack-tracker-wii.dll_unloaded, version: 0.0.0.0, time stamp: 0x5f3e3852
Exception code: 0xc0000005
Fault offset: 0x000d9b32
Faulting process id: 0x2924
Faulting application start time: 0x01d684332970ab83
Faulting application path: C:\Users\lfend\Downloads\opentrack-2.3.12-portable\install\opentrack.exe
Faulting module path: opentrack-tracker-wii.dll
Report Id: 723de5a7-59ac-4554-9ea0-997cf9784149
Faulting package full name: 
Faulting package-relative application ID: 

This doesn't interfere with my usage of OT.

kristankenney, 2020-09-20:
I can confirm this issue happening as well with OpenTrack 2.3.12 when starting tracking, the previous version 2.3.11 seems unaffected. This error also appears when exiting OpenTrack.

sthalik, 2020-09-21:
Duplicate of #1139. If this is a problem for you then delete `opentrack-tracker-wii.dll`.

closed 2020-09-21 by sthalik