opentrack/opentrack #1159: Crashes when I click Start Tracking

markallisongit, 2020-09-07:
Program crashes when I click start tracking with and without **Preview** enabled. This is a new Windows 10 installation.

Using opentrack-2.3.12 and have UDP over network as input.
Windows firewall disabled.

Log:
```
[2020-09-07 09:26:00.870] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-07 09:26:01.312] [aitrack] [info] Created/Found prefs.ini
[2020-09-07 09:26:01.312] [aitrack] [info] Searching for cameras...
[2020-09-07 09:26:02.659] [aitrack] [info] Number of recognized cameras: 1
[2020-09-07 09:26:02.661] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2020-09-07 09:26:02.661] [aitrack] [info] Building Tracker with selected camera: 0
[2020-09-07 09:26:02.706] [aitrack] [info] Tracker initialized.
[2020-09-07 09:26:02.706] [aitrack] [info] Updated stabilizer.
[2020-09-07 09:26:02.706] [aitrack] [info] Updating camera parameters...
[2020-09-07 09:26:02.706] [aitrack] [info] Updated camera parameters. 640x480@0
[2020-09-07 09:26:02.706] [aitrack] [info] App initialized
[2020-09-07 09:26:04.635] [aitrack] [info] Saving prefs
[2020-09-07 09:26:04.635] [aitrack] [info] Updated stabilizer.
[2020-09-07 09:26:04.635] [aitrack] [info] Updating camera parameters...
[2020-09-07 09:26:04.635] [aitrack] [info] Updated camera parameters. 640x480@0
[2020-09-07 09:26:04.635] [aitrack] [info] Tracker initialized.
[2020-09-07 09:26:04.636] [aitrack] [info] Prefs saved
[2020-09-07 09:26:05.875] [aitrack] [info] Starting camera 0 capture
[2020-09-07 09:26:06.258] [aitrack] [info] Camera 0 started capturing

```

sthalik, 2020-09-07:
Is it aitrack or opentrack crashing? aitrack is a third-party project.

markallisongit, 2020-09-07:
I've been playing around with the settings and noticed that FPS was set to 0 by default. I changed that to 24 and it doesn't crash now. What's the best FPS setting?

markallisongit, 2020-09-07:
> Is it aitrack or opentrack crashing? aitrack is a third-party project.

Sorry I posted to the wrong GitHub

closed 2020-09-07 by markallisongit