opentrack/opentrack #1164: Hamilton filter glitches

LiftPizzas, 2020-09-09:
When using the Hamilton filter I get occasional repeated skipping back and forth between two 180 degree different views. (For instance the roll will twitch at 0 and 180 several times a second.)

It is most common when my yaw and pitch are near zero. I can supply video if needed.

GO63-samara, 2020-09-09:
Yes, show video of this problem, please.

LiftPizzas, 2020-09-09:
Video linked. Problem shown with and without tracker screens. If you watch the Roll graph/position you can see it's wrapping to 180.
https://youtu.be/mDlfIRWKJgQ

I can reduce the effect by making Roll output 0 always, but there's still some twitching happening. Seems like an "ignore all inputs larger than" (an inverse deadzone?) might be super helpful. :)

GO63-samara, 2020-09-09:
In the video, on the "Raw tracker data" panel, you can see jumps along the Roll, Z axes, for example, at 00:22, 00:25, 00:32 and others. This data has not yet been processed by the filter, it comes directly from the tracker.
Therefore, the problem is not with the Hamilton filter, but with your tracker, which works over UDP over network (AI track?).
I noticed that the Roll axis is rotated 90 degrees, it may need centering.
Good luck!