opentrack/opentrack #1166: extreme game data swings

problematicman, 2020-09-10:
Hello!

I'm using a delanclip, a modified ps3 eye camera, and opentrack 2.3.12. Whenever I turn my head with just yaw or pitch, all the other axes seem to respond and zip back and forth in the x and z axis. I'm using the profile from https://forum.il2sturmovik.com/topic/34403-a-complete-guide-to-set-up-head-tracking-opentrack/ , but changing the profile didn't seem to help, as well as the filter.

Heres a video of the issue: https://youtu.be/XD1N-5rw3dk

Thanks!

sthalik, 2020-09-15:
That looks like effects of relative translation. Do you have it enabled?

problematicman, 2020-09-16:
Disabling it seems to have worked, but now the head movements are backwards when i turn around. Is there a way I can still have the realistic movement?