opentrack/opentrack #1168: support for Audeze Mobius

nactrum, 2020-09-14:
Audeze Mobius headphones includes their own headtracker. Can you add a support for it to opentrack? I dunno which protocol they use,

magiricod, 2020-09-24:
I too am interested in this have been looking for a solution to plug in the roll yaw and pitch data

QUAKEULUS, 2021-03-03:
interested in this as well. the values are displayed in the mobius/hyperx software, maybe a hook could be written to read those?

Slion, 2021-03-03:
Nice headphones, hope you guys find someone interested in implementing this.

QUAKEULUS, 2021-03-03:
It's very possible.
I have both hyperx and audeze (same software, different builds/skins) how can i assist?

[image]

Slion, 2021-03-04:
I'm sure it is possible, you just need someone with the hardware that has the time and skills to do it.

Neo-Grotesk, 2021-09-25:
I've just bought a set of Audeze Mobius headphones, from the tracking in the Audeze app it looks like it would be perfect for using with Opentrack. I have no idea how you go about getting the tracking data from Audeze to talk to Opentrack, but I'm happy to try if there's some sort of guide on how to get started creating something like that. Probably way beyond my capabilities but happy to take a look if anyone has any info as a starting point for someone completely uneducated!

Thiewin, 2023-10-27:
Hello, have you finally found the solution? I have a mobius too.

QUAKEULUS, 2023-10-27:
I didn't get around to this, but I can lend a developer a mobius if they'd implement it.

Thiewin, 2023-10-27:
That would be awesome!