opentrack/opentrack #1171: request: emulate Tobii Eyetracker

Scx8R5SvSlYB3xcM0psn, 2020-09-18:
It would be really nice to emulate the Tobii eye tracker by Yaw/Pitch movement of the head. The rest of axis could be ignored.
Why? There are some games, that doesn't support head tracking but instead eye tracking. This would give more potential use cases for Opentrack. 

For example some recent titles: _The Division 2_ and _Shadow of The Tomb Raider_. For more, see the Tobii website.
I would greatly appreciate the possibility to be the beta tester.

sthalik, 2020-09-28:
See: #1037.

closed 2020-09-28 by sthalik

Scx8R5SvSlYB3xcM0psn, 2020-10-09:
@sthalik I think this is the other way: requested that with Opentrack you would emulate the existence of the Tobii Eye Tracker. When you move your head, the movement (yaw, pitch) is then translated as were you looking at specific area on your screen.

I have the feeling, that the mentioned #1037 is to use Tobii device for head tracking. The opposite.

sthalik, 2020-10-17:
OK, then the question is whether the Tobii API allows for exposing third-party devices. But eye trackers are really different from head trackers and I don't believe you'd fine it working fine ingame.