opentrack/opentrack #1172: how to build for x-plane output

phiber2k4, 2020-09-19:


coljac, 2021-11-06:
This should be considered a bug, as the readme says that X-Plane plugin is an output mode, but as @phiber2k4 points out it does not show up in the output list.

After looking in the source, and following a hunch, I finally set the output to Wine and then it did work in X-Plane. I don't know quite why or what is happening here technically, but it did work. Can anyone confirm there's not supposed to be an X-Plane output?

sthalik, 2021-11-06:
It's using the Wine shared memory mapping for simplicity's sake. For that reason there's no duplicate output. Furthermore, you can build the Wine output without Wine itself, for that reason.