opentrack/opentrack #1174: TrackClip hat vertices

whiskeygreg, 2020-09-22:
Hello again!

A friend of mine got a 3d printer and I had him print me this: 
[image]

From what I understand, this is the same thing as the official one. Unfortunately, I can't seem to get the vertices correct. I'm getting a lot of jitter when I move my head to the right, sometimes even an extreme jump in position, and stuttery tracking on the other side. 

[image]
[image]

I'm sure I'm just making a silly mistake somewhere in here, but I can't seem to pinpoint it. Any help would be appreciated.

Slion, 2020-09-22:
There is a nasty bug in the current release.
Try one of those instead:
https://slions.net/threads/open-track-for-kinect-v2.67/

whiskeygreg, 2020-09-22:
Hello again, Slion! I am using that build, since I'm the one that spurred you to make it. Thanks again for that!

I believe this is an issue with my vertices. I've been tinkering with the numbers to try and hone in on the problem, but can't seem to nail it down.

Slion, 2020-09-22:
Also note that at some point I realised that the default TrackIR hat is not best for EasyTracker you can get better results with a wider custom hat as seen in the docs and other issues here.

whiskeygreg, 2020-09-22:
I will have my friend print me a bigger clip, and if that doesn't work out I'll move on to a more custom solution like I had before.

Thanks for your continued support!

Slion, 2020-09-23:
I think your vertices definition is messed up too. See: https://slions.net/threads/open-track-for-kinect-v2.67/

You may want to reduce the deadzone for more responsiveness.

whiskeygreg, 2020-09-23:
It doesn't track if the Top is 0,0,0. I'll keep tinkering.

Slion, 2020-09-24:
> It doesn't track if the Top is 0,0,0. I'll keep tinkering.

Fixed the last link above. Here it is again https://github.com/opentrack/opentrack/issues/1141#issuecomment-685762072
It explains how to set your vertices pretty thoroughly.

Slion, 2020-09-24:
I've also slightly improved Easy Tracker docs by adding information from that post.
https://github.com/opentrack/opentrack/wiki/Easy-Tracker#model-coordinate-system

MathijsG, 2020-10-17:
Care to share the STL file?

Slion, 2020-10-19:
> Care to share the STL file?

What's that?

MathijsG, 2020-10-19:
> > Care to share the STL file?
> 
> What's that?

I mean the 3D print file I see in the first post.

Slion, 2020-10-19:
> I mean the 3D print file I see in the first post.

For me this:
https://3dmdb.com/en/3d-model/TrackClip-hat-for-Trackir-by-astfede/5029049/?q=trackir