opentrack/opentrack #1177: opentrack non-stop stops tracking

ProrokC2, 2020-09-26:


ProrokC2, 2020-09-27:
I FINALLY MADE IT WORK.
The camera used to freeze constantly, it was completely unusable.
I uninstalled the CL driver, I stopped craping around with zadig and driver changes. 
I uninstalled all the previous software, including CLeye and downloaded a different driver called iPi. And now it works out of the box! No problems what so ever. Leaving it here, maybe someone will need it someday, just like I did.

ProrokC2, 2020-09-27:
Ok I figured some stuff out. Cleaned the drivers, the camera works and zandig driver is permament. The camera freezes when I plug my joystick in. And I have no idea why. I left it tracking aruco for 10 minutes and works flawlessly, i plug my joystick in and it freezes in an instant. My joystick is t.flight hotas x, windows 10 home 64 bit, camera installed via libusbK, joystick has standard drivers.

ProrokC2, 2020-09-28:
switching the usb port fixed the issue

closed 2020-09-28 by ProrokC2

sthalik, 2020-09-28:
How's your user experience? Is the CPU usage lower? Do you get any freezes or crashes?

ProrokC2, 2020-10-15:
No freezes, no crashes. User experience is stellar, works as intended