opentrack/opentrack #1179: PS3Eye Camera glitchy with libusb

Diddo17, 2020-10-06:
Hey,
I updated opentrack and I saw that it added support for the "PS3Eye open driver".
Well as I have been using the CodeLaboratory Driver I wanted to switch to libusb anyway so I though yeah, why not.
So I downloaded Zadig and changed the driver to libusb and it worked without any error.

But now the camerafeed is like this:
[image]

It was fine with the codelaboratory driver, no glitches whatsoever, but since the change to libusb it looks like that.
I also tried Zadig again to switch to libusbk but that looks exactly the same.

I have a three-point homemade tracking hat, if thats relevant in any way.
Can someone help me?

Thanks :)

Diddo17, 2020-10-06:
Nevermind. 
My old .ini was totally fucked with fucked everything with it.
If you have this issue too, change to a different .ini and you should be fine.

I'll close this myself.

Diddo17, 2020-10-06:
Sorry for this nightmare of a thread, never really used the github issues function before.

sthalik, 2020-10-08:
That's normal. Above 75 Hz the image is partially corrupted

Diddo17, 2020-10-08:
Ah, alright then I'm relieved to hear that I didnt screw something up.

closed 2020-10-08 by Diddo17

sthalik, 2020-10-17:
Can you compare the CL-Eye and open driver full system CPU usage? Thanks.

Diddo17, 2020-10-21:
This is the CPU usage over 60 seconds for the CL-Driver:
[image]

This is the CPU usage over 60 seconds for the Open Driver:
[image]

In both tests I moved my head more than usual but everything worked just fine.
I didnt have the artifacts on the CL-Driver if I went over 83hz though, something that I still have with the Open Driver (although it doesnt bother me because the tracking works fine).

I would've attached the .blg files but they arent supported and I dont know how to reach you privately.

sthalik, 2020-10-25:
Thank you, this is important feedback. It's good that you can reproduce the reduced CPU usage.

> I didnt have the artifacts on the CL-Driver if I went over 83hz though, something that I still have with the Open Driver

The CL-Eye driver doesn't allow for the 83 Hz mode. 75 Hz has mild artifacts as normal, while 60 Hz has none.