opentrack/opentrack #1183: ver 2.3.10 looks better than ver 2.3.12

Stix-09, 2020-10-16:
Just updated to ver 2.3.12 and noticed the size of the opentrack  gui is slightly bigger and fonts look worse compared to the previous version 2.3.10 I had installed, I tested the portable and installed ver 2.3.12 and both have this issue.

I know its a minor thing, but the older version looks better

[image]

[image]

sthalik, 2020-10-16:
Are you using HiDPI?

sthalik, 2020-10-17:
Run opentrack with either of command-line arguments (best through Windows console), see if any of them help:

- `opentrack.exe -platform windows:dpiawareness=0`
- `opentrack.exe -platform windows:dpiawareness=1`

Stix-09, 2020-10-17:
Thanks for looking at this.
Ok I tried the above , results are the same for all. no change.
The display looks clear ie there is no blurring (I read about dpi this on docs.microsoft.com) 
Looks to me like its using a font that is not present. 
I downloaded the portable versions and also tested them.
Results are same as installed versions.
this is what they look like running side by side 
As you can see the window size is also dif between versions and font also looks dif.
[image]

Stix-09, 2020-10-17:
It looks clearer than above, (crisp) prob just the jpg compression. Looking at above just now, I think font is the same, just the windows size, between 2 versions is different , almost like window was stretched between old and new versions to to a larger size.

GO63-samara, 2020-10-17:
I have the same problem. My main window looks exactly the same in versions 2.3.12 and 2.3.11. The switches "-platform windows: dpiawareness = 0-2" do not help. Version 2.3.10 looked significantly better. I think that in extreme versions Bold fonts are used, and the labels do not fit within the borders of the controls. In addition, in version 2.3.10 the pose-widget width was equal to the width of the "Game data" panel and it looked much better than in version 2.3.12. Hoping for the best!

sthalik, 2022-09-30:
Can you confirm this with [opentrack 2022.3.0](https://github.com/opentrack/opentrack/releases/tag/opentrack-2022.3.0)?

GO63-samara, 2022-09-30:
Since OT-2.3.13 the windows look right for me, as in OT-2.3.10
In OT-2022.3.0 the windows look correct for me too.

closed 2022-09-30 by sthalik