opentrack/opentrack #1185: curve mapping issue.

Iknowlongnamerightwhatwasithinking, 2020-10-23:
The output of the curve is still outputting something even when the curve in the curve mapping is zero. 
This introduces a tiny bit of UNDEAD zone.... 

ie. If you turn off all but X. and set the curve to have a massive deadzone. 
You will still get a tiny bit of undeadzone around zero. 
You cannot see this in the "game data" BUT you can see the octypuss fidget. 
( make sure you are crossing zero to see the fidget ) 

this comes through loud and clear in game. !!! 

Seems to only do it on X.

2.3.12 and 2.3.10 do this. 

[image]

So if I hover around ZERO I get an in game fidget. Which you can see on the octypussy. 

Thanks! 

Me. xxx

sthalik, 2020-10-25:
Add a point at zero, then it won't happen anymore.

Iknowlongnamerightwhatwasithinking, 2020-10-25:
Ok will try that. Will it be fixed in the future? ( the other axis don't have this issue...) 
Thanks!  xxx

Iknowlongnamerightwhatwasithinking, 2020-10-25:
Your suggestion does not work, sorry!
Even with a point placed at zero the curve is still making the X move like a bump around zero. 

What I expect is nothing to happen till I X to a certain point then it tracks the curve.
The fidget around the zero becomes a distraction when zoomed in. 

Sorry to bring this to your attention! 

Me.