opentrack/opentrack #1187: Could not find SteamVR in input section after build

kadogo, 2020-10-25:
Hello,

My system is Debian Buster I wanted to use SteamVR as input so I first tried to follow the wiki (https://github.com/opentrack/opentrack/wiki/Building-on-linux). The build went without any issue but when I start opentracker I don't see "SteamVR".

I think it's just not build, I add the cmake command output in case it can help. I not see "steamvr" in the module section so I guess it's not found by cmake but I don't know why.

```
$ cmake .
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- User settings file 'sdk-paths-user@GNU-Linux.cmake' doesn't exist
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- module api
-- module compat
-- module csv
-- module cv
-- module filter-accela
-- module filter-ewma
-- module filter-hamilton
-- module user-interface
-- module logic
-- module migration
-- module executable
-- module options
-- module pose-widget
-- module proto-fgfs
-- module proto-libevdev
-- module proto-udp
-- module qxt-mini
-- module spline
-- module tracker-freepie-udp
-- module tracker-fusion
-- module tracker-linux-joystick
-- module tracker-pt-base
-- module tracker-pt
-- module tracker-s2bot
-- module tracker-test
-- module tracker-udp
-- module video-opencv
-- module video
-- -- Compile flags:
--  CXX: -ggdb -Wall -Wextra -Wpedantic
--  CXX_RELEASE: -O3 -march=native
-- -- Link flags:
--  LINK_SHARED: 
--  LINK_SHARED_RELEASE: 
-- -- Static archive flags:
--  STATIC: 
--  STATIC_RELEASE: 
-- --
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/GITS/opentrack
```

If you need any other output let me know I will provide them.

Have a good day.
Cheers

kadogo, 2020-10-26:
I went a bit further.

*   I added `otr_module(tracker-steamvr)` at first line in ./tracker-steamvr/CMakeLists.txt

*  I copy the openvr.h files to the tracker-steamvr directory because I didn't know how add a library path

```
cp /usr/include/openvr/openvr* ./tracker-steamvr/
```

*  I ran the same command as before

```
cmake .
make
make install
```

But when I launch `./install/bin/opentrack` I have the next error

```
DEBUG [/home/user/GITS/opentrack/api/plugin-support.hpp:107]: library "steamvr" failed: "Cannot load library /home/user/GITS/opentrack/install/bin/../libexec/opentrack//opentrack-tracker-steamvr.so: (/home/user/GITS/opentrack/install/bin/../libexec/opentrack//opentrack-tracker-steamvr.so: undefined symbol: VR_GetInitToken)"
```

kadogo, 2020-10-27:
I do another message because it stay clearer than edit.
By changing the CMakeList I have fixed the Issue and I just need to launch SteamVR first like it's said on https://github.com/opentrack/opentrack/issues/1118

I guess the main issue is that the openvr lib in not properly find but I'm not good enough to know how to fix that properly so I will just add what I did. Don"t hesitate to ping me to make more test if needed.

```
--- tracker-steamvr/CMakeLists.txt	2020-10-27 22:57:39.594148475 +0100
+++ tracker-steamvr/CMakeLists.txt	2020-10-27 22:57:35.417991222 +0100
@@ -31,16 +31,9 @@
     set(steamvr-dll "libopenvr_api.dylib")
 endif()
 
-if(steamvr-dll AND opentrack-intel)
-    if(steamvr-lib STREQUAL "")
-        set(steamvr-lib "${steamvr-dll}")
-    endif()
     SET(SDK_VALVE_STEAMVR "" CACHE PATH "Valve's SteamVR")
-    if(SDK_VALVE_STEAMVR)
         otr_module(tracker-steamvr)
-        install(FILES "${SDK_VALVE_STEAMVR}/bin/${steamvr-dir}/${steamvr-dll}" DESTINATION "${opentrack-libexec}")
+        install(FILES "/usr/lib/x86_64-linux-gnu/libopenvr_api.so" DESTINATION "${opentrack-libexec}")
 
-        target_include_directories(opentrack-tracker-steamvr SYSTEM PRIVATE "${SDK_VALVE_STEAMVR}/headers")
-        target_link_libraries(opentrack-tracker-steamvr "${SDK_VALVE_STEAMVR}/lib/${steamvr-dir}/${steamvr-lib}")
-    endif()
-endif()
+        target_include_directories(opentrack-tracker-steamvr SYSTEM PRIVATE "/usr/include/openvr")
+        target_link_libraries(opentrack-tracker-steamvr "/usr/lib/x86_64-linux-gnu/libopenvr_api.so")
```

The evdev protocol is making something weird (the yaw when I look at left/right is making my mouse move up / down instead of left / right). But I will try to build proto-vjoystick that is not build something for something similar than tracker-steamvr.

sthalik, 2020-11-02:
So it looks like the build system doesn't support system openvr. What happens if you use out-of-tree steamvr as on Windows? The Valve repo has binary releases and that's what I compile-tested on Linux.

kadogo, 2020-11-02:
Hello @sthalik 

To be honest I didn't do much compilation an my english is not my first language so I not really know the terms like "out-of-tree" but if there are commands or links to guide me I can try them.

I will try to find some information about it to try to advance but if you have ressources or precise commands to try don't hesitate to ping me I will answer so fast as I can.

sthalik, 2020-11-02:
Clone the repo from https://github.com/ValveSoftware/openvr, it already has binaries. Point out SDK_STEAMVR to that directory.

kadogo, 2020-11-02:
Hello

I did the next command in my opentracker directory

```
$ git clone https://github.com/ValveSoftware/openvr
# Edit tracker-steamvr/CMakeLists.txt for pointing SDK_VALVE_STEAMVR (I didn't found only SDK_STEAMVR)

$ git diff
diff --git a/tracker-steamvr/CMakeLists.txt b/tracker-steamvr/CMakeLists.txt
index eb8b20b9..6b7258d2 100644
--- a/tracker-steamvr/CMakeLists.txt
+++ b/tracker-steamvr/CMakeLists.txt
@@ -35,7 +35,7 @@ if(steamvr-dll AND opentrack-intel)
     if(steamvr-lib STREQUAL "")
         set(steamvr-lib "${steamvr-dll}")
     endif()
-    SET(SDK_VALVE_STEAMVR "" CACHE PATH "Valve's SteamVR")
+    SET(SDK_VALVE_STEAMVR "/home/user/GITS/opentracker/openvr" CACHE PATH "Valve's SteamVR")
     if(SDK_VALVE_STEAMVR)
         otr_module(tracker-steamvr)
         install(FILES "${SDK_VALVE_STEAMVR}/bin/${steamvr-dir}/${steamvr-dll}" DESTINATION "${opentrack-libexec}")

$ cmake .
-- User settings file 'sdk-paths-user@GNU-Linux.cmake' doesn't exist
-- module api
-- module compat
-- module csv
-- module cv
-- module filter-accela
-- module filter-ewma
-- module filter-hamilton
-- module user-interface
-- module logic
-- module migration
-- module executable
-- module options
-- module pose-widget
-- module proto-fgfs
-- module proto-libevdev
-- module proto-udp
-- module qxt-mini
-- module spline
-- module tracker-freepie-udp
-- module tracker-fusion
-- module tracker-linux-joystick
-- module tracker-pt-base
-- module tracker-pt
-- module tracker-s2bot
-- module tracker-test
-- module tracker-udp
-- module video-opencv
-- module video
-- -- Compile flags:
--  CXX: -ggdb -Wall -Wextra -Wpedantic
--  CXX_RELEASE: -O3 -march=native
-- -- Link flags:
--  LINK_SHARED: 
--  LINK_SHARED_RELEASE: 
-- -- Static archive flags:
--  STATIC: 
--  STATIC_RELEASE: 
-- --
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/GITS/opentrack
```

I think it doesn't find it but maybe I'm point out SDK_VALVE_STEAMVR in a wrong way.

sthalik, 2020-11-02:
Use `ccmake`.

kadogo, 2020-11-02:
I didn't know about ccmake thanks. It looks like it's working same if I have an error when it compile the steamvr tracker

```
cmake .
ccmake .
# Edit SDK_VALVE_STEAMVR with the path of the openvr repo
# configure and generate in ccmake
cmake .
# I saw steamvr in the modules
make
```

```
[  1%] Built target i18n-module-video
[  1%] Built target i18n-module-api
[  7%] Built target i18n-module-compat
[  7%] Built target i18n-module-csv
[  8%] Built target i18n-module-cv
[  9%] Built target i18n-module-filter-accela
[  9%] Built target i18n-module-filter-ewma
[ 10%] Built target i18n-module-filter-hamilton
[ 12%] Built target i18n-module-user-interface
[ 14%] Built target i18n-module-logic
[ 14%] Built target i18n-module-migration
[ 15%] Built target i18n-module-executable
[ 18%] Built target i18n-module-options
[ 19%] Built target i18n-module-pose-widget
[ 19%] Built target i18n-module-proto-fgfs
[ 20%] Built target i18n-module-proto-libevdev
[ 20%] Built target i18n-module-proto-udp
[ 21%] Built target i18n-module-qxt-mini
[ 22%] Built target i18n-module-spline
[ 23%] Built target i18n-module-tracker-freepie-udp
[ 23%] Built target i18n-module-tracker-fusion
[ 24%] Built target i18n-module-tracker-linux-joystick
[ 25%] Built target i18n-module-tracker-pt-base
[ 26%] Built target i18n-module-tracker-pt
[ 26%] Built target i18n-module-tracker-s2bot
[ 27%] Built target i18n-module-tracker-steamvr
[ 27%] Built target i18n-module-tracker-test
[ 28%] Built target i18n-module-tracker-udp
[ 28%] Built target i18n-module-video-opencv
[ 28%] Built target i18n-lupdate
[ 28%] Built target i18n-qm-nl_NL
[ 28%] Built target i18n-qm-zh_CN
[ 28%] Built target i18n-qm-stub
[ 29%] Built target i18n-qm-ru_RU
[ 29%] Built target i18n-lrelease
[ 29%] Built target i18n
[ 30%] Built target opentrack-version
[ 31%] Built target opentrack-api
[ 44%] Built target opentrack-compat
[ 50%] Built target opentrack-options
[ 51%] Built target opentrack-csv
[ 52%] Built target opentrack-video
Scanning dependencies of target opentrack-cv
[ 52%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/affine.cpp.o
[ 52%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/init.cpp.o
[ 52%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/translation-calibrator.cpp.o
[ 53%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/video-widget.cpp.o
[ 53%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/moc_affine.cpp.o
[ 53%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/moc_init.cpp.o
[ 53%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/moc_numeric.cpp.o
[ 53%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/moc_translation-calibrator.cpp.o
[ 54%] Building CXX object cv/CMakeFiles/opentrack-cv.dir/moc_video-widget.cpp.o
[ 54%] Linking CXX static library opentrack-cv.a
[ 55%] Built target opentrack-cv
[ 57%] Built target opentrack-spline
Scanning dependencies of target opentrack-filter-accela
[ 57%] Building CXX object filter-accela/CMakeFiles/opentrack-filter-accela.dir/ftnoir_filter_accela.cpp.o
[... 95 lines cut ...]
Scanning dependencies of target opentrack-tracker-pt
[ 91%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/camera.cpp.o
[ 91%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/frame.cpp.o
[ 91%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/module.cpp.o
[ 92%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/point_extractor.cpp.o
[ 92%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/moc_camera.cpp.o
[ 92%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/moc_export.cpp.o
[ 92%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/moc_frame.cpp.o
[ 92%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/moc_module.cpp.o
[ 92%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/moc_point_extractor.cpp.o
[ 93%] Building CXX object tracker-pt/module/CMakeFiles/opentrack-tracker-pt.dir/qrc_tracker_pt.cpp.o
[ 93%] Linking CXX shared library opentrack-tracker-pt.so
[ 94%] Built target opentrack-tracker-pt
[ 95%] Built target opentrack-tracker-s2bot
[ 95%] Building CXX object tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/steamvr.cpp.o
/home/user/GITS/opentrack/tracker-steamvr/steamvr.cpp: In lambda function:
/home/user/GITS/opentrack/tracker-steamvr/steamvr.cpp:257:20: error: ?class vr::IVRSystem? has no member named ?ResetSeatedZeroPose?
                 v->ResetSeatedZeroPose();
                    ^~~~~~~~~~~~~~~~~~~
make[2]: *** [tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/build.make:76: tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/steamvr.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2976: tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

I tested it on the unstable branch and opentrack-2.3.12 tag it's the same issue.

sthalik, 2020-11-02:
Replace `ResetSeatedZeroPose` with `ResetZeroPose(vr::TrackingUniverseSeated);`.

kadogo, 2020-11-02:
I did the replace and launch `make` again and now I have this error

```
/home/user/GITS/opentrack/tracker-steamvr/steamvr.cpp: In lambda function:
/home/user/GITS/opentrack/tracker-steamvr/steamvr.cpp:257:20: error: ?class vr::IVRSystem? has no member named ?ResetZeroPose?
                 v->ResetZeroPose(vr::TrackingUniverseSeated);
```

toby3d, 2022-05-18:
Faced similar problems on the commit db54d8eb86b06 The default build does not include the `tracker-steamvr` module.

Providing the `SDK_VALVE_STEAMVR` variable to the directory with the downloaded https://github.com/ValveSoftware/openvr repository and trying to build it will cause errors:
```bash
/home/toby3d/git/opentrack/tracker-steamvr/steamvr.cpp: In lambda function:
/home/toby3d/git/opentrack/tracker-steamvr/steamvr.cpp:257:20: error: ?class vr::IVRSystem? has no member named ?ResetSeatedZeroPose?
  257 |                 v->ResetZeroPose(vr::TrackingUniverseSeated);
      |                    ^~~~~~~~~~~~~
make[2]: *** [tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/build.make:136: tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/steamvr.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2823: tracker-steamvr/CMakeFiles/opentrack-tracker-steamvr.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

toby3d, 2022-05-18:
> Please see whether this patch works with the current openvr version.

Okay, now the build is successful and the corresponding menu item is now present in Data Input list. But when I try to start the server the following error occurs:

[image]

sthalik, 2022-05-18:
@toby3d Can you confirm that openvr works in other apps?

toby3d, 2022-05-18:
> Can you confirm that openvr works in other apps?

I'm not sure, I haven't used SteamVR before. How can I check?

sthalik, 2022-09-30:
> I'm not sure, I haven't used SteamVR before. How can I check?

I don't even have the device. One surefire way to get it to work is installing Windows.