opentrack/opentrack #1188: PS3 Eye Camera Basic Headtracking (No LED)

Richdem, 2020-10-27:
Is there a way to use the PS3 Eye Camera for headtracking with this excellent software? I tried originally with the Smooth Track app on my phone but didn't like the battery drain so I picked up a cheap PS3 Eye Camera. I know it will work with a modded camera and led clip but that is a project for another day. For now just want to use head tracking

Thanks in advance

Len62, 2020-10-27:
You can try using the Aruco filter with opentrack. You have to print out a marker, figure a way to mount it, and away you go. Here's a link to a short video on using the Aruco filter: https://www.youtube.com/watch?v=w6CCzYy5hRs

Richdem, 2020-10-28:
> You can try using the Aruco filter with opentrack. You have to print out a marker, figure a way to mount it, and away you go. Here's a link to a short video on using the Aruco filter: https://www.youtube.com/watch?v=w6CCzYy5hRs
> There are other videos out there, just google "opentrack with Aruco filter".

Thank you ,

This is brilliant. I quickly tried this last night and I was amazed. Way better than video head tracking even with a large square stuck to my head lol. I am going to mount to a cap today and test further. Think I'll be ordering an LED clip soon though

gconry18, 2020-12-18:
@Richdem, Another option is https://github.com/AIRLegend/aitrack which I use a lot.

Richdem, 2020-12-18:
@gconry18 , Thanks for this I will take a look.

I did end up getting a full LED head tracker from Grassmonkey solutions and it is awesome, Cannot believe I waited this long to try head tracking lol