opentrack/opentrack #1190: Creating AppImage on Linux, Error loading opentrack-tracker-pt.so when executing

paju1986, 2020-11-01:
Hello.

I successfully created an AppImage for Linux compiling from source, the application runs fine but it's missing point tracker.
When i execute the appimage from terminal i get the following errors:

DEBUG [/home/alberto/proyectos/opentrack/api/plugin-support.hpp:107]: library "easy" failed: "Cannot load library /tmp/.mount_opentreZkCmh/usr/bin/../libexec/opentrack//opentrack-tracker-easy.so: (opentrack-video.so: no se puede abrir el fichero del objeto compartido: No existe el fichero o el directorio)"
DEBUG [/home/alberto/proyectos/opentrack/api/plugin-support.hpp:107]: library "pt" failed: "Cannot load library /tmp/.mount_opentreZkCmh/usr/bin/../libexec/opentrack//opentrack-tracker-pt.so: (opentrack-video.so: no se puede abrir el fichero del objeto compartido: No existe el fichero o el directorio)"
DEBUG [/home/alberto/proyectos/opentrack/api/plugin-support.hpp:107]: library "opencv" failed: "Cannot load library /tmp/.mount_opentreZkCmh/usr/bin/../libexec/opentrack//opentrack-video-opencv.so: (opentrack-video.so: no se puede abrir el fichero del objeto compartido: No existe el fichero o el directorio)"

It appears than it has trouble loading those share libraries from the libexec forlder inside the appimage, but if i copy the libexec forlder into my /usr/libexec on my system, them they load fine. Also i realize that in the path there is a double "/", maybe that has something to do with it??

Regards!!

paju1986, 2020-11-01:
here is the appimage: [https://www.pling.com/p/1440394/startdownload?file_id=1604176591&file_name=opentrack-x86_64.AppImage&file_type=application/x-executable&file_size=112160744](url)

sthalik, 2020-11-02:
Use `LD_DEBUG`.

paju1986, 2020-11-02:
Executed with LD_DEBUG=libs, output log.

[debug.txt](https://github.com/opentrack/opentrack/files/5476363/debug.txt)

paju1986, 2020-11-02:
executed with LD_DEBUG=libs

[log.txt](https://github.com/opentrack/opentrack/files/5476392/log.txt)

sthalik, 2020-11-02:
Link statically with opencv.

paju1986, 2020-11-03:
I have no idea how to do that, do i have to set something on the makefile?

sthalik, 2020-11-03:
Set with `ccmake` when building opencv.

paju1986, 2020-11-03:
I fixed it!! 
i had to modify opentrack-library-path.hxx and change the directory for looking to the libs to empty string
then moved all the libraries from libexec to the bin directory and the remake the appimage with linuxdeploy
Now it starts without error and i get all the Inputs!!
The updated appimage is here:

[https://www.pling.com/p/1440394/startdownload?file_id=1604425245&file_name=opentrack-x86_64.AppImage&file_type=application/x-executable&file_size=216817640](url)

sthalik, 2020-11-03:
That's not the correct solution.

paju1986, 2020-11-03:
i know its not the perfect way to do it, but im not very familiar with cpp programing and i dont know how to do it better. But my target was to create an appimage for linux and i did it and it works perfectly so i guess this is solved for me.

closed 2020-11-04 by paju1986