opentrack/opentrack #1191: P3Dv4.5 & OpenTrack

charco92, 2020-11-03:
Hi everyone, 

I know this topic has been discussed before, but unfortunately I was not able to fix my problem just by reading from the previous threads. 
I have installed the latest version of OpenTrack (and Smooth Track on my mobile device) but P3Dv4.5 doesn't seem to recognise it at all. 
I know this has to do with the 32bit DLL vs. 64bit architecture of P3Dv4 and SimConnect, but did anyone managed to make it work ? 
If so, could you please me help out ? 

Take care, 
Thomas.

sthalik, 2020-11-03:
> I know this has to do with the 32bit DLL vs. 64bit architecture of P3Dv4 and SimConnect,

Not really, no.

What's the P3D SDK for 4.5? Can you send me simconnect.dll?

charco92, 2020-11-04:
Hi, 

So I've downloaded that one http://cloud.prepar3d.com/SDK/Prepar3D_v4_SDK_Setup_4.5.14.34698.msi but I've also installed the legacy version which were included in the root folder of P3D. 
Which dll do you want ? All of them ? 

Thanks a lot

charco92, 2020-11-09:
Hi, 

Any chances for an update or help ? :) That would be grand. I am still struggling to get it working on P3D...! 

Thanks a lot, 
Thomas

charco92, 2020-11-16:
Problem solved by myself.

closed 2020-11-16 by charco92

sthalik, 2020-11-16:
Explain it for others please.