opentrack/opentrack #1192: Track stuttering / camera FPS oscillating

Limper1111, 2020-11-07:
Hi, yesterday when playing DCS and using Opentrack, this problem developed during game and lasts to this moment. The problem is that my cockpit view started to stutter, move unnaturally. I recorded the video, so you can see what is happening exactly.
Tried unplugging and replugging the joystick, pedals, camera, but nothing helped. The thing I noticed before playing was the sound when you unplug the USB, but everything looked fine, the problem developed during playing.
Thanks in advance for help.
https://youtu.be/NQf6K516Eog

plule, 2020-11-07:
Hi, I have the exact same issue since yesterday too: FPS of the PS3 camera is oscillating and the acelera filter becomes crazy. Maybe a Windows update broke something?

I've noticed that another filter (Hamilton? Forgot the name) was not affected

Limper1111, 2020-11-07:
Hi, I partially solved it. In PS3 CL eye test I found this setting, set it to 75 fps, and ticked the box.
[image]

Symerizer, 2020-11-08:
I have the same problem too, since yesterday. Other filters are not as bad as Accela, but they all stutter a bit.

I've also found something in the Point Tracker 1.1 settings: [Camera FPS](https://gyazo.com/3123bb45a335e8068fb5bd2d5d6fd7af)

The frame rate of the camera itself seems to stutter. I don't know if that was like this before.

Cosmo-Cave, 2020-11-08:
Same issue. Limper1111 fix didn't work for me.

Cosmo-Cave, 2020-11-08:
plule's fix worked for me. I didn't have the Hamilton filter at first because I hadn't updated to the newest version of Opentrack.

GO63-samara, 2020-11-08:
The "Hamilton filter" can be added to Opentrack-2.3.11. Just copy opentrack-filter-hamilton.dll to the "modules" folder.
https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-hamilton-filter
https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-download/opentrack-filter-hamilton.dll

sthalik, 2020-11-10:
Can you see if it's any better with the open driver?

plule, 2020-11-11:
I have the fluctuating fps issue with both the open driver and the other, and no matter what’s in the CLEye configuration. Actually, I even see it with another unrelated logitech webcam. Whatever the configuration is the displayed fps always fluctuate between 0 and the selected fps.

I could not find any relevant change on Windows, the only one matching the timeline is some antivirus definition update that is basically done every day...

Limper1111, 2020-11-11:
Hi, my Opentrack is now without issues. As I mentioned before the thing that repaired it was setting FPS in all settings at 75 (in CL eye test and in Opentrack). Also I removed the IR filter from PS3 camera so the tracking points are now a lot bigger. FPS are almost stable at 75 now.

plule, 2020-11-11:
On my side I went berserk, removed all possibly related drivers, including usb stuff, and after reinstallation it’s also back to steady 75 Hz. I’m trying to mess around in Cl-eye test and opentrack to recreate the issue with no luck. So really not sure what happened, but it works too now.

JBoysen1, 2020-11-13:
I also had the FPS of a PS3Eye camera fluctuate between single digit fps and 75 leading to stuttering. Similar to video by Limper1111 above. I noticed the fps count in Pointtracker1.1 settings page. Tried installing latest Opentrack (2.3.12) but that made no difference, camera fps still all over the place. And of course, if you have 2 fps it will only read your head position twice in a second so stuttering and view jumping must be coming from that.

Only thing that worked was to reinstall camera drivers, I used the PS3 driver package mentioned here at Delanclip where I bought my tracking system: https://delanengineering.com/manual/. Reinstalled, restarted and now have stable fps with Accela filter.

Gilbo26, 2021-01-03:
Also having this problem with my PS3EYE. Soon as I start CL-EYE Test, or start tracking in Opentrack, not only does my fps stutter similarly as described above, but it also causes my mouse and keyboard to lag. I dont get the sound or notification of a usb device disconnecting or reconnecting but that's what it appears as. I tried uninstalling all my usb contollers and drivers, etc, fresh reinstall, but that didn't do the trick. Kind of at a loss here. I'm on the most current version of windows 20H2, but I think this started a day ago after a security update? It'd been working fine up until Jan 1st. Though I uninstalled any recent windows updates and had no luck there either.

plule, 2021-01-03:
It just reappeared today for me too, it was OK 2-3 days ago.

Windows 10 1909, the only update listed is the security antivirus one... Same symptoms as before, FPS is fluctuating in opentrack whatever the driver is, breaking the accelera filter.

I don’t get anything strange on mouse or keyboard though.

Gilbo26, 2021-01-03:
@plule , I might try and fresh install 1909 since I know it was working on that. Its such a weird issue, and it has to be driver or usb related but something must be conflicting. I haven't yet ruled out the camera itself yet though, so I'll try it on another pc and see if I have issues or not. Don't think thats the problem, though i'm not sure of the behavior of a dying ps3 eye.

On another note, I thought it may be the difference between usb 3.0 and usb 2.0 ports, as I only have 3.0 ports, but since it had been working for the longest time on 3.0 ports I'm led to believe that for me thats not the problem. Plus I tested what I believe to be a 2.0 extension cable, which should have by-passed the usb 3.0 wires/leads rendering the port a 2.0 port, but that didn't fix it either. I'll update when I try another pc, but if it isn't the camera itself, my best guess is a windows update is causing some odd driver confliction or usb issue. From there i'll do a fresh install back at win 10 1909 and cross my fingers.

kavdborr, 2021-01-12:
I'm having the same problem. I also plotted the yaw and dt for the Accela filter and it looks quite nasty:
[image]

Overall the application responds much slower then it used to, taking dozens of seconds to open from the desktop and it hangs when I press the start button...

Why do all nice things need to be ruined by a windows update...

Gilbo26, 2021-01-13:
@kavdborr I recreated the issue on a third PC by updating to all the latest windows updates under 20H2, same issue happened with all applications using the ps3 eye camera/driver. After rolling back to 1909, and 1 restart later it was fixed. So its 100% something in the windows update. All we can do is wait for the next update and hope that whatever is conflicting is fixed. Most likely something to do with compatibility of older windows (7 or 8) drivers.

kavdborr, 2021-01-15:
@Gilbo26 The problem was fixed with the latest windows updates for me. Now I'm on version 20H2 with KB 4598242.

gakun17, 2021-01-19:
Same exact issues. Limper1111's fix didn't work. Symerizer's footage of the frame rate going wild also happens. It began out of nowhere while I was playing American Truck Simulator and tracking suddenly got very uncomfortably jittery and I couldn't fix it since.

Gilbo26, 2021-06-30:
@kavdborr Issue as reappeared on 20H2 19042.1052. Unfortunately I can no longer roll back to a previous version. Its a sad day. Will report back if I find anything new fix wise

sthalik, 2021-07-01:
Interesting. What happens if you set opentrack to realtime priority?

GO63-samara, 2021-07-02:
As a hypothesis. Track stuttering appears only for some users. Perhaps track stuttering is associated with an FFB joystick? There are certain problems with Direct Input and FFB. Here is about this and the solution to replace the dll (in Russian):
[https://forum.il2sturmovik.ru/topic/14042-ил2-битва-за-форс-фидбек-ms-ffb2trackir/](url)

Manateekida, 2022-08-22:
Hey so i had the exact same Problem right now! Yesterday i was playing rts2 with my old xbox360 for PCs controller and tracking went all fine. Today i turned my pc on, startet opentrack and bleh, stuttering 2, 22, 75, 2, 22, 75, 2,... fps. Remembered that i had a problem once with my controller plugged in so i unplugged it and started opentrack again, bleh. But guess what, very smart me thought maybe restarting without the controler helps? And voila, everything runs smooth again! Even with my controller pluged in after turning my PC on ;)

I hope that helps someone! I´m just writing it because i already had a github acc :D