opentrack/opentrack #1199: Turn off all deadzones if 1 deadzone is exceeded

Mikestriken, 2020-12-04:
Hello, I was wondering if there is a way to remove all deadzones you create, should one deadzone be exceeded.
Explanation, the deadzones for the most part seem to be there so that you can look straight ahead without twitchy movement, however, if you yaw slightly to the right (to the point where you exceed the yaw's deadzone), you probably no longer wish to have a deadzone for pitch.

If this were added it would add a whole new level of complexity (I can imagine) as you would have to create multiple graphs for each axis to represent what each graph would look like without a deadzone and this may result in a sudden jerk after you leave a deadzone when each axis gets adjusted to the new graph; however I believe that with enough tweaking one could get past that issue.

Stratium, 2025-02-01:
Bumping this feature request. Current dead zone options work as below
[image]

It would be nice if this was an option instead
[image]

Mikestriken, 2025-02-01:
What I ended up doing is just turning off the deadzones entirely lol. I made all my rotation and movement linear and then just got used to it.