opentrack/opentrack #1201: Trying to build with SDK_WINE: ON causing build issue.

mkoreneff, 2020-12-08:
Running Ubuntu 20.04.
I have both i386 and amd64 winehq-staging versions installed via winehq ppa. 

I can build without any issues following the instructions on the wiki (thank you!)

When trying to build with `SDK_WINE: ON` i'm facing an issue which is most likely due to my wine environment defaulting to see 64bit libraries. Does anyone know how to make wine look at 32bit libraries when building? 

```
/usr/bin/ld: relocatable linking with relocations from format elf64-x86-64 (/opt/wine-staging/lib64/wine/libwinecrt0.a(exe_entry.o)) to format elf32-i386 (opentrack-wrapper-wine.mcDCZn.o) is not supported
winebuild: /usr/bin/ld failed with status 1
winegcc: /opt/wine-staging/bin/winebuild failed
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:82: proto-wine/opentrack-wrapper-wine.exe.so] Error 2
make[1]: *** [CMakeFiles/Makefile2:2458: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
What does building with SDK_WINE give me ontop of the normal output? Will I be able to integrate with games easier?

sthalik, 2020-12-08:
> When trying to build with SDK_WINE: ON i'm facing an issue which is most likely due to my wine environment defaulting to see 64bit libraries. Does anyone know how to make wine look at 32bit libraries when building?

`winegcc -m32` needs to work.

mkoreneff, 2020-12-08:
Thanks for the quick response! Sorry for the rudimentary questions.

How can I confirm that winegcc -m32 is working? 
Calling winegcc -m32 shows the following
```
/media/local/opentrack$ winegcc -m32
gcc: fatal error: no input files
compilation terminated.
winegcc: /usr/bin/gcc failed
```

I can see -m32 is passed correctly from `proto-wine/CMakeFiles/wine-wrapper.dir/build.make`, however the result is as above.
```        cd /media/local/opentrack/proto-wine && wineg++ -mconsole -g -DNOMINMAX -O2 -m32 -std=c++17 -fPIC -o opentrack-wrapper-wine.exe -I /media/local/opentrack -I /media/local/opentrack /media/local/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx /media/local/opentrack/proto-wine/opentrack-wrapper-wine-posix.cxx /media/local/opentrack/proto-wine/opentrack-wrapper-wine-windows.cxx -Wall -Wextra -Wpedantic -lrt
```

sthalik, 2020-12-08:
You're missing wine-tools for 32-bit wine. See:

```sh
echo 'int main(void){return 0;}' | winegcc -m32 -xc -
```

mkoreneff, 2020-12-08:
> You're missing wine-tools for 32-bit wine. See:
>
> echo 'int main(void){return 0;}' | winegcc -m32 -xc -

Thanks for the tip. I managed to compile after installing wine32-tools
`sudo apt install wine32-tools`

The-Foilist, 2023-04-14:
I'm having the exact same error, but the error persisted after I installed wine32-tools.