opentrack/opentrack #1204: Permanent drift after long session, issues cause off axis rotation.

YellowOnion, 2020-12-15:
[image]

looking down, raw data is correct, but BOTH pitch and yaw are influenced.

[image]

looking up, same thing but different direction.

[image]

I suspect this is due to gimbal lock / fine rounding errors accumulating over time, I took off my headphones and layed them down for 30mins.

sthalik, 2020-12-15:
This is inertial sensor drift, which is unavoidable.

YellowOnion, 2020-12-15:
The raw data isn't drifting it's the output, look at pitch, it only changes by 2 degrees, but BOTH pitch and roll have increased, I only have 2x multiplier in my mappings.

Even if it's drift, it should be fixed by the recentre button, but it's not.

sthalik, 2020-12-16:
> Even if it's drift, it should be fixed by the recentre button, but it's not.

What happens when you center?

GO63-samara, 2020-12-16:
I would advise to disable "Center at startup" and disable filters for testing.

YellowOnion, 2020-12-16:
> 
> 
> > Even if it's drift, it should be fixed by the recentre button, but it's not.
> 
> What happens when you center?

center is only valid for a forward look, when I pitch up it rolls to the left, and when I pitch down, it rolls to the right.

I'm also getting this on a fresh start (roll and pitch are swapped for no reason: 
[image]

perhaps related to issue #1200  ?

sthalik, 2020-12-17:
How would you guys feel if there was a center mode that merely subtracted values with no respect to each other?

YellowOnion, 2021-08-30:
Sorry I didn't reply to this, I honestly not sure if that would fix the issue...

Upon thinking about this, it seems to me that pitch and roll are applied relative to the raw  yaw at zero, as in the pitch and roll are pitched and rolled as if at raw yaw 0 is "forward".

If I face south and apply +180 to input correction it seems to disappear mostly.