opentrack/opentrack #1205: SmoothTrack Android App with OpenTrack and Vjoystick Outputs Not Detected by Monitor

sontran20, 2020-12-17:
Good evening everyone,

I am in desperate need of your technical expertise. I have the SmoothTrack app for my Android cell phone which uses the phone to track head movements as input through UDP over network. When I choose Output as freetrack 2.0 Enhanced everything works perfectly in Digital Combat Simulator, MS Flight Simulator 2020, etc. in tracking head movements wirelessly. However when I switch Output to Joystick Emulation -- vjoystick my head movement input is shown with the moving Octopus but when I open Monitor vJoy either nothing or a very tiny small jump in axis is detected. I installed the newest version then uninstalled using Add / Remove Programs and tried with the old version but nothing works. I'm trying to get this working for my favorite game Ace Combat 7 on Steam. I tried the Steam guide but couldn't get it to work since I use SmoothTrack instead of TrackIR in the guide. Any help would be greatly appreciate! 

Newest: https://sourceforge.net/projects/vjoystick/files/Beta%202.x/2.1.9.1-160719/
Old: https://sourceforge.net/projects/vjoystick/files/Beta%202.x/2.1.8.39-270518/
SmoothTrack App: https://smoothtrack.app/ 
https://steamcommunity.com/sharedfiles/filedetails/?id=2195724313

sontran20, 2020-12-17:
Good morning everyone, 

I figured out the issue and was able to get OpenTrack to send Output to Joystick Emulation -- vjoystick. For anyone else having this issue what I did was open vJoyConf and click ResetAll. This ensures only 1 controller. Then I typed in the values in the linked screenshot from the Ace Combat 7 Steam Guide and clicked Apply. When I went to vJoyMonitor my head movements were all sent wireless from my cell phone through SmoothTrack to Open Track and then to vJoy. There was no need to run any script in FreePIE. I'll close this issue for now and continue to experimenet with x360ce to see if everything comes together. 

[Ace Combat vJoyConf](https://steamuserimages-a.akamaihd.net/ugc/1490082212996646138/FEE409BFE9F0EDE40BDB5E6067FFF68FDCC72769/)

closed 2020-12-17 by sontran20

cartablanca08, 2024-02-08:
> Good morning everyone,
>
> I figured out the issue and was able to get OpenTrack to send Output to Joystick Emulation -- vjoystick. For anyone else having this issue what I did was open vJoyConf and [...]

hi there, i'm trying to replicate your set up, but this time, i'm using AITrack as an input. the octopus move around following my head movement but, i have the exact same problem now and tried your workaround to no avail. i've reinstalled vjoy 2 times now. and yes, i hope i can use it for AC7, bind the vjoy with x360ce.

sontran20, 2024-02-08:
Thank you very much for reaching out. Unfortunately vjoy is a real pain to
use and very buggy. The same goes with x360ce. I will never use either of
them again. My solution worked great for me at the time however I stopped
playing AC7 for a few months. When I came back vjoy was once again useless.
I installed and uninstalled it like 10x to no avail. I don't know if it's
related but the next day my computer completely died. I had a new power
supply I had only used for a few months and my guestimate the motherboard
was dead. I had to sell all my PC components. Corona hit and I couldn't
build a PC again until last year.

I recently got back to testing AC7. This time with a VR headset called the
Bigscreen Beyond although the solution works for Quest 2/3 as well. I use a
mod called Universal Unreal Injector (UEVR) by PrayDog to play AC7 in VR.
There are some limitations such as the clouds not working properly and
using 3rd person camera due to cockpit issues but it is a way better
experience then headtracking ever provided. I know it's the pricier option
but time to setup, stability, and experience in VR is very immersive.