opentrack/opentrack #1207: I wrote an iOS app for headtracking by mounting iPhone to head

Cecemel20, 2020-12-19:
Hi, I had seen apps for android and windows phone that could do this, but not for iOS so I wrote one.
Link to the GitHub repo: https://github.com/Cecemel20/iOSHeadTrack
One little annoyance is the installation process since the app isn't on the App Store as I'd have to pay 99 euros/year for that.

Interested in hearing your thoughts. You might want to consider adding it to the tracking input list in the readme.

gconry18, 2020-12-20:
@Cecemel20 does this integrate with opentrack?

Cecemel20, 2020-12-21:
@gconry18 yes, it sends the phone's attitude via UDP to opentrack's 'UDP over network' input. Instructions are in the 'usage' section of the readme.