opentrack/opentrack #1209: Can you add faceAPI to opentrack?

slavsquatter3000, 2020-12-23:
Is it possible to add faceAPI (the face tracker from the payware HeadtrackNoIR) to open track?

Tequiller, 2021-04-18:
Same claim :)

Should be nice to keep same game parameters (mapping) and to be able to play either night (with IR Leds) or day without the necessity to close shutters :)

Thank you for all your great work !

Tequiller, 2021-04-18:
faceAPI seem to be free:

https://sourceforge.net/p/facetracknoir/wiki/faceAPI/

slavsquatter3000, 2021-04-18:
> Same claim :)
> 
> Should be nice to keep same game parameters (mapping) and to be able to play either night (with IR Leds) or day without the necessity to close shutters :)
> 
> Thank you for all your great work!

?, dont get what your saying here

FoldZero, 2021-06-20:
Im wondering the same thing. FaceTN is quite outdated now,would like to see OPENTRACK Support it as an INPUT

FoldZero, 2021-06-20:
> faceAPI seem to be free:
> 
> https://sourceforge.net/p/facetracknoir/wiki/faceAPI/

But how would one integrate it?