opentrack/opentrack #1210: "Toggle" function doesn't work anymore, Z value keeps increasing

hoover67, 2020-12-23:
I just updated to 2.3.12 on win10 / 32 bit in the hope it would fix the issue I'm having with the "toggle" function for the last few months.

Previously, when hitting toggle the view would freeze at the point I was looking at until I hit the toggle key the next time. 

Now, the z axis value keeps increasing until it hits roughly 10,000 and then just sits there until the next time. The octopus jumps far back in the preview and wiggles up and down continuously (as far as I can make it out). 

Any idea what could be causing this? Using a ps/eye with pointtracker plugin, as mentioned the feature worked fine before. 

thanks & all the best, 

Uwe

hoover67, 2020-12-23:
[image]

hoover67, 2020-12-23:
this is the view with the "toggle" key pressed once.

JBoysen1, 2020-12-23:
I also suspect a bug with Toggle (not the Toggle Tracking). After pressing the key mapped to Toggle, the raw tracker data stop moving as expected. But the game data keep on changing, in my case the Z value instantly skips to a fairly large negative value (approximately -900) and the Y value counts rapidly up to 9955. All other 4 values are unchanged.

I can see this happening directly in the Opentrack app, in game it of course causes all sorts of problems. But this can be replicated without any game active

hoover67, 2020-12-24:
The weird thing is this bug crept out out of nowhere without me changing
anything in the setup as far as I can remember.

Thanks for your confirmation!

All the best,

Uwe

>

hoover67, 2020-12-24:
I recompiled opentrack (latest git sources) on Linux Mint 20 and the toggle bug has gone or at least it's not present in the "octopus test" view.

JBoysen1, 2020-12-24:
Interesting, seems there is a fix already in place in the code. Might try and see what it takes to recompile from latest source on Windows

hoover67, 2020-12-24:
Let me know if the latest source works on win10 as well and please share the binary build :) 

All the best, uwe

Eraysor, 2020-12-24:
Would love to see a fix for this!

JBoysen1, 2020-12-31:
I got my hands on another profile which solved the problem. So good news is that the binary that I run was ok, the problem was the profile. Bad news is that I have tried a lot of things to make my old profile work based on differences between it and the other profile which works. But I could not figure out what the extra feature the other profile has which made that profile work and my old one fail.

asotelles, 2021-01-14:
its when you change the "custom center position". If you put any value there the toggle button does that anomaly. I'm searthing for a fix but couldn't find yet

Eraysor, 2021-03-13:
Not fixed in today's update sadly...

syswpit, 2021-03-19:
I can confirm this has to do with the customer center position. Even if you just change the pitch by 0.1 degree, the toggle function breaks.
This is even without any input, and regardless of input/output or filter, so I think the issues lies within OpenTrack itself.
I hope it gets fixed soon.

GO63-samara, 2021-04-25:
Fixed operation of the **[Toggle]** and **[Toggle while held]** buttons: https://github.com/opentrack/opentrack/commit/0ddcf4eb94fa4ee530702bca2c54e390b3c6057d
 
Try replacing `opentrack-logic.dll` in **Opentrack-2.3.13** with dll from this archive:
https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-download/FixToggleButton.rar

Good luck!

Eraysor, 2021-04-26:
> Fixed operation of the **[Toggle]** and **[Toggle while held]** buttons: [0ddcf4e](https://github.com/opentrack/opentrack/commit/0ddcf4eb94fa4ee530702bca2c54e390b3c6057d)
> 
> Try replacing `opentrack-logic.dll` in **Opentrack-2.3.13** with dll from this archive:
> https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-download/FixToggleButton.rar
> 
> Good luck!

Thank you! Worked for me.

asotelles, 2021-05-03:
for me didnt work, opentrack did not runned with this dll, but thx anyways

zmz125000, 2021-08-16:
FixToggleButton.rar
https://sites.google.com/site/diyheadtracking/
https://drive.google.com/file/d/1-FASGQdVXVONO3_ULGlDHGZeD7GKL_-e/view

sthalik, 2021-09-29:
It's been fixed in 2021.1.0.

closed 2021-09-29 by sthalik