opentrack/opentrack #1213: [Feature request]

MathijsG, 2021-01-04:
Hey Stanislaw,

I was wondering where the max in-game CM of 75 is based on? Could it be a possible feature to have make this adjustable? I see some games that could do more than 75 in-game CM's.

sthalik, 2021-01-05:
> I was wondering where the max in-game CM of 75 is based on? 

Back then there was no combobox allowing you to select the maximum output value. So adding more output made using the plot more inaccurate.

> Could it be a possible feature to have make this adjustable?

Yes, that makes sense. It's a 10-minute change, but I don't have the environment set up and it's going to be a while, honestly.

MathijsG, 2021-01-06:
Thanks Stanislaw,

Would be great if we could set these values in the future. Best would be to let it be user input?

For the physical Z-zoom axis the default selection options for the amounts of CM also doesn't always fit people's usecases either. Maybe we should make this configurable as well?

sthalik, 2021-01-06:
> Would be great if we could set these values in the future. Best would be to let it be user input?

I'm normally against things being *too* configurable. Anyway, what's the range of values you're interested in?

> Maybe I could change this myself, I'm unemployed next month :-P.

That's sad about your employment. Feel free to send a pull request.

I should release 2.3.13 but that needs the Wii tracker to be fixed first.

MathijsG, 2021-01-06:
I agree with things not always need to be too configurable, but this is one of the things where the use case can be so broad, and every game has a different amount of zoom, that's why I opt for every reasonable value via user input.

I don't know if there is a hard limit on the TrackIR protocol side of things? My use case with this feature was DCS, where I could just move 75 CM forward, while I could zoom in much further with another non-tracking axis. Maybe limit it to 999 CM max as user input?

sthalik, 2021-01-07:
> I don't know if there is a hard limit on the TrackIR protocol side of things?

No, it's just a value. Games clamp it though. Maybe install the TrackIR software? I'm not going to accept its EULA so can't do it myself.

MathijsG, 2021-01-07:
Checked it with my TrackIR which I bought for testing purposes. Doesn't look like Z is really limited. It goes to 300+ CM's. So a 3 digit value as max seems fine to me?

I've uploaded the sgl.dat to: https://mega.nz/file/DVQBTKLI#5iw3yB1Vpg8D496WmR-R9GUqQ26ifqrAfS7LjoTClm4

MathijsG, 2021-09-29:
Could this be implemented easily?

sthalik, 2022-04-26:
Yes.

closed 2022-04-26 by sthalik