opentrack/opentrack #1217: Only works for a minute

GaryMack, 2021-01-11:
Hi,
Firstly, it's a great product!
It works very well but after a minute or so, I get a "no route to host" error on the phone app. If I hit stop and start on the phone app it will continue to work fine for another minute, then the cycle continues.
Something seems to be timing out.
Data logging doesn't show anything. I've tried multiple profiles, even downloaded ones, just in case.
Any ideas on how to fault find this?

Windows 10 and iPhone X

Thanks in advance

christopher-gray-vca, 2021-01-16:
Hey, I'm having the same issue, I have mine set up on my Mac and it works until after a few mins then it freezes.

GaryMack, 2021-01-17:
Hi,
I fixed my by turning off the phone VPN completely. It shouldn't have been active as my home WiFi was 'trusted' but hey, it worked.
Good luck

closed 2021-01-17 by GaryMack