opentrack/opentrack #1218: Output Stuttering

crunchywaffle, 2021-01-12:
https://user-images.githubusercontent.com/47266709/104268662-46141880-5449-11eb-9dee-eeea6cc640ae.mp4

Like the video shows, recently OpenTrack has been going cuckoo with its output. I'm using opentrack-2.3.12 here. This profile is using all default settings, running on UDP over network, outputting using freetrack 2.0 enhanced, and with the Accela filter. 

The other filters have the exact same problem, although Hamilton seems to be affected differently; it stutters with tiny movements but much more rapidly than the other filters, I think it might be because I was testing it with maxed out smoothing.

At first I thought it was my input(I'm using an app on my phone that uses facial recognition to recognize movement), but as you can see in the video, the raw tracker data correctly shows me turning my head from side to side, but the game data flips out.

This issue only showed up recently, < 1 week of this post, but it persists after a PC restart and even after I deleted and redownloaded Opentrack.

GO63-samara, 2021-01-12:
In my opinion this is the same problem as here:
https://github.com/opentrack/opentrack/issues/1192
It looks like it is related to the latest Windows update. We need to wait for the next Windows update.

crunchywaffle, 2021-01-15:
As of Windows version 1909, this has been fixed.

closed 2021-01-15 by crunchywaffle

mlb406, 2021-03-07:
I believe this problem still persists. My PC is on version 2004 of Windows 10 , and yet I'm having the exact same issue as @crunchywaffle describes in the above post. The issue started today, it was fine two days ago, and I did not use my PC at all yesterday. I don't believe there have been any changes to my system since. I have been on windows 10 verison 2004 since February 4 so I don't think it is a problem related to this, nor do I have the option of rolling back to check.