opentrack/opentrack #1220: Opentrack exe wont start??

griffmeister84, 2021-01-13:
Hi 

Ive downloaded opentrack to use for smooth track. I click on the OT exe, and wont start up?? I’m using Windows 8 64 bit.  Tried the compatibility troubleshooter and no luck.  Any help would be much appreciated.

egk696, 2021-01-15:
I am facing the same problem and have posted the following issue https://github.com/opentrack/opentrack/issues/1212 but no luck so far.

griffmeister84, 2021-01-15:
I feel it might be a windows update that has interfered?

John can you give any update to our problem?

Rgds

Tommy

epaga, 2021-01-18:
Hey guys, saw I was tagged on this - to be clear I'm not the author of OpenTrack, I'm the author of [SmoothTrack](https://smoothtrack.app), which can send data to OpenTrack, that's all.

FWIW, this hasn't happened to me before - something that HAS happened to me though is I once activated to have OpenTrack be in the system tray, so I thought it wasn't starting up when in reality it was in the system tray. Could this be what is going on? I turned that option off and it then started working normally again.

robert914, 2022-01-29:
I'm seeing this same problem starting `opentrack.exe` on Windows 10 (64bit) using the latest release 2022.1.1 (also had the issue with 2022.1.0 prior). It doesn't always happen, but the executable starts and I can see it running in the Task Manager, but the window never appears, and nothing appears in the task bar either. Sometimes it will eventually appear after waiting about 1-2 minutes, but other times it doesn't seem to open.

robert914, 2022-01-29:
I may have found what was causing my issue ... maybe. Have to see if it continues to work, but I uninstalled all the Logitech Camera software I had installed (Logitune, Logitech Capture, etc.). So it may be that there was some conflict with the 10 or so processes it had running. Now OpenTrack opens without issue, at least for now.... time will tell

sthalik, 2022-02-24:
Thanks for confirming this issue. For what it's worth, you can switch the Logitech webcam to a generic UVC driver with no loss in functionality.

Player4532, 2022-03-06:
> Hey guys, saw I was tagged on this - to be clear I'm not the author of OpenTrack, I'm the author of [SmoothTrack](https://smoothtrack.app), which can send data to OpenTrack, that's all.
>
> FWIW, this [...]
Thanks you so much that was exaktly my problem.

Saky1146, 2022-05-23:
I have exactly the same problem, crashes on startup without any UI. No window. Appreciate any guidance.

Saky1146, 2022-05-24:
Endorse @Player4532 reply. Same thing happened to me. I thought Opentrack is not starting, but on checking my system tray after reading his reply, relieved to find that it was there. Opentrack has an option to disable this minimisation. So no major software issue. Thanks 4532.

DogBadger, 2023-09-07:
I know this is over a year since the last comment, so if I am not supposed to post here, let me know. So to my issue. Windows 10, 36GB Mem, I7 system.  I had been  using opentrack for about a week or so then all of a sudden it stopped working. I noticed in task manager there where about 8 instances of it running so I ended those and tried to restart it. Only one instance starts up but nothing is launched on screen. Uninstalled it then reinstalled it. Same result. I have no Logitech SW installed. It is very odd. Any thoughts?