opentrack/opentrack #1222: Head tracking not working with ATS

firegod381, 2021-01-18:
I have been trying to get opentrack to work with ATS for over 2 weeks now. I have changed the config code to trackIR “1” I have uninstalled both ATS and Opentrack. I am using a self made clip with diodes that when I pull up opentrack and Cl eye test work perfectly. However when I open ATS it does not work and only lets me use my mouse to move the camera. I am at a lose on what to do. I have been reading and searching for ways to fix it so I can use opentrack but I have not been able to get it to works to fix my problem.  

And help would be greatly appreciated.

pilpo, 2021-01-18:
Hello, the game wasn't detected for me in Opentrack (on the top bar of opentrack "default.ini:: Unknown game". I installed Opentrack on C:\ instead of I:\ and launched the game. Everything is fine for me now. Maybe try that :)

firegod381, 2021-01-18:
Thank you for the suggestion, I tried both of them and it is still not working.

firegod381, 2021-01-19:
I finally got it fixed! What I had to do to fix it in case anyone else has the problem. First I uninstalled opentrack, cl eye test and ATS. Than I installed in this order, ATS, cl eye test and than opentrack. Once all were installed went into ATS and could not move the camera with mouse so opened opentrack and started tracking and could move the screen with my head.

Nerezza, 2021-08-28:
This may or may not be the actual solution to this, but I just spent 2 hours trying to get Opentrack to work with ATS and this issue was a common result in Google, so I want to potentially save others a lot of time and/or frustration by leaving my results here:

ETS2 and ATS both have an option to disable TrackIR integration (which is used by Opentrack ofc), but this option isn't available in the in-game options. If you navigate to Documents > American Truck Simulator or Euro Truck Simulator 2 depending on the game you're playing, find "config.cfg" (NOT the one under the steam_profiles folder) and edit it in a text editor.

Find the line: uset g_trackir
Make sure it reads: uset g_trackir "1"

This is the option to enable or disable trackir, but for whatever reason it's not present in the in-game options. I don't know how mine got disabled, but it took me 2 hours to find out this option existed and it was disabled.