opentrack/opentrack #1228: need release done

sthalik, 2021-01-29:
Hey,

@Slion @GO63-samara @cpuwolf 

Can any of you do a release? The only thing that needs to be done is the `tracker-wii` .dll not loading due to some static initializer. There's the `msvc.cmake` toolchain file.

Can you do the official release this time, please? I can't do anything, I'm sorry 👎 

sh

sthalik, 2021-01-29:
At the moment I'm unable to do any work on opentrack. But the current state of the stable release is disrespectful toward great work @cpuwolf did on `tracker/wii`.

GO63-samara, 2021-01-29:
Hi!
I manage to build only part of the libraries, opentrack-tracker-wii.dll is not among them.
But!
The WiiPointTracker 1.1 sources haven't been updated since 2019, have they?
WiiPointTracker 1.1 is not listed as trackers in Opentrack-2.3.12. However, if you replace opentrack-tracker-wii.dll (1,835,008 bytes) with the library from OT-2.3.11 (1,562,112 bytes), then WiiPointTracker 1.1 appears in the tracker list.
Maybe this will help?

dAlexis2, 2021-01-31:
Yes, maybe mothly releases from master branch are needed as unstable. I see tobii inside but gathering all needed 4 building...

sthalik, 2021-01-31:
> I manage to build only part of the libraries, opentrack-tracker-wii.dll is not among them.

Hey,

See `sdk-paths-sthalik-MSVC-windows` and `cmake/msvc.cmake`, then <<https://github.com/opentrack/opentrack-depends>>. Not all dependencies have been included due to licensing issues. Use MSVC Preview 2019 or something similar, compile all with `cmake/msvc.cmake`.

If you can do that, you can have the commit bit and do releases with no questions asked -- basing upon your previous contributions.

The static initializer needs fixing and the Tobii tracker isn't critical for the new release.

I can still mentor on various parts of the codebase (especially the core library parts) and maintain most of it.

GO63-samara, 2021-01-31:
Thank you, this is useful information. But I was building via cmake-gui.

The file "sdk-paths-sthalik@MSVC-windows.cmake" contains the following lines:
```
...
setq (Qt5_DIR "../qt-5.12.0/lib/cmake/Qt5")
...
set (Qt5_DIR "c: /qt/qt-5.14.0/lib/cmake/Qt5" CACHE PATH "" FORCE)
...
```
There is some kind of contradiction here.

Also I am having difficulties connecting Eigen3. I can't fix the Eigen3_DIR-NOTFOUND error.
How do I compile and install Eigen3 correctly?

sthalik, 2021-01-31:
> Also I am having difficulties connecting Eigen3. I can't fix the Eigen3_DIR-NOTFOUND error.

 Try `ninja install` on Eigen.

> There is some kind of contradiction here.

It's a copy-paste error. The last statement takes precedence since it's `CACHE FORCE`.

> How do I compile and install Eigen3 correctly?

You don't compile it or link to it, it's header-only.

Just make sure to verify the amount of files in the `modules/` install prefix subdirectory against the last of my releases. And remember about the toolchain file.

There's also the part of a custom qtbase/qttools/qtserialport with edited mkspecs, but the only important part about Qt is building it with `-MT` and the same MSVC as the rest of the dependencies. Edit `mkspecs/win32-msvc.conf` and `mkspecs/common/msvc-*.conf`, it's the only way to be sure. It takes not that long to build Qt if you use only qtbase/tools/serialport.

If you REALLY don't want to build Qt, since it's the biggest PITA of them all, some random recent Qt is good enough for public use.

GO63-samara, 2021-02-01:
"You don't compile it or link to it, it's header-only."

Yes it worked!
I renamed "sdk-paths-sthalik@MSVC-windows.cmake" to "sdk-paths-Win7@MSVC-windows.cmake" and only then did this file connect.
In "sdk-paths-Win7@MSVC-windows.cmake" I changed the paths to mine:

- setq (Qt5_DIR ...
- set (Qt5_DIR ...
- setq (OpenCV_DIR ...

And I put the Eigen sources in the "opentrack-depends" folder.

I had to delete folders that required the SDK, or did not find header files, I will deal with them later:

- kinect-face
- rift-140
- rs
- steamvr

As a result, MSVC built Opentrack. But compared to my past experience, only the Kalman filter has appeared. The rest of the trackers and output protocols did not appear.

I'll figure it out tomorrow.

[image]

[image]

[image]

sthalik, 2021-02-02:
You need to put in all opentrack-depends stuff, LibOVR, libusb, and stuff that I couldn't legally redistribute. Subdirectories have `CMakeLists.txt` with details and generally it's `SDK_*` that needs to be filled. Note that if you fill one `SDK_` or `foo_DIR` then you have to reconfigure and see if there are no more new `SDK_*` variables waiting to be filled.

Also use `CMAKE_TOOLCHAIN_FILE` as `cmake/msvc.cmake` at the time when first configuring. I recommend executng `cmake-gui` from `cmd` after enabling `vcvars32.bat` and selecting Ninja or NMAKE. It should configure itself as MSVC on its own.

When dealing with Qt, make sure the install directory of Qt itself isn't in PATH, otherwise it might not work for users. Other than that, opentrack's install script should automatically copy everything necessary from the Qt dir.

I build opencv static (with the toolchain file, as everything else) with everything except DSHOW disabled and static CRT.

It takes about an hour to set up if you already know how to do that, but it's a PITA the first time.

The important part is enabling catching win32 SEH exceptions and fixing the `tracker-wii` initializer.

GO63-samara, 2021-02-09:
I am still unable to link OpenCV to Opentrack.
I built OpenCV like this:
Chose Win32 in cmake-gui. Clicked [Configure].
I put the following options in cmake-gui:
```
BUILD_SHARED_LIBS     OFF
WITH_DSHOW            OFF
BUILD_WITH_STATIC_CRT ON
```
Again clicked [Configure].

As a result, I got the following flags:
```
CMAKE_CXX_FLAGS_DEBUG		/MTd /Zi /Ob0 /Od /RTC1
CMAKE_CXX_FLAGS_MINSIZEREL	/MT /O1 /Ob1 /DNDEBUG
CMAKE_CXX_FLAGS_RELEASE		/MT /O2 /Ob2 /DNDEBUG
CMAKE_CXX_FLAGS_RELWITHDEBINFO	/MT /Zi /O2 /Ob1 /DNDEBUG
```
Clicked [Generate], [Open Project]. Everything is OK.
In MSVC 2019 built a solution for Debug, then for Release, then Install. Everything is OK.

I have a few questions about the build sequence in MSVC:

1. Whether it is necessary to build all three targets (Debug, Release, Install). What targets does Opentrack need?
2. To build Opentrack do I only need *.lib and *.pdb library files or something else?

sthalik, 2021-02-10:
> I am still unable to link OpenCV to Opentrack.

Build it, no need to install. Then specify `OpenCV_DIR` to the build directory (not the source directory).

> As a result, I got the following flags

Just remember to use opentrack's toolchain file for the final build.

> To build Opentrack do I only need *.lib and *.pdb library files or something else?

`.lib`, yes. `.pdb` are installed only if they exist.

Slion, 2021-02-15:
I'm afraid I can't invest much time on that either. I never actually did a proper release. Can it even be done on Windows 10?

sthalik, 2021-02-15:
> I never actually did a proper release. Can it even be done on Windows 10?

Yes, with no worries.

> I think I usually build the x64 variant for some reason but the releases are still x86 right?

Yes, the release has no reason to use amd64.

> If I recall well I think Qt was a pain to setup for x86. Could be wrong though been months since I did anything there.

Generally it's a pain until you figure it out for the first time.

Slion, 2021-02-15:
I actually documented my issues with x86 there:
https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows#optional-build-dependencies
Apparently it's that there was no x86 OpenCV distributions.

GO63-samara, 2021-02-16:
Hi guys!

Thanks for the help!
I was helped a lot by your answers and also issue https://github.com/opentrack/opentrack/issues/861
Only today I finally managed to link OpenCV and OpenTrack.
And I managed to build a Release with the /MT and /MTd flags for Win32, partially.
For today I received the following:

Trackers:

- Easy Tracker 1.0
- FreePie UDP receiver
- Fusion
- Hatire Arduino
- Joystick input
- PointTracer 1.1
- Razer Hydra -- Inertial device
- S2Bot receiver
- Test tracker
- UDP over network
- WiiPointTracker 1.1

Output protocols:

- FlightGear
- Freetrack 2.0 Enchanced
- FSUIPC -- Microsoft FS2002/FS2004
- Joystick emulation -- vjoystick
- Microsoft FSX SimConnect
- mouse emulation
- UDP over network

Filters:

- Accela
- EWMA
- Hamilton
- Kalman

The rest of the parts require additional SDKs, I'll deal with them tomorrow.

This is how I did it using only CMake-gui (3.17.0), QT-5.14.1 and MSVC2019.
In CMake-gui in the first [Configure] I set:
```
[Configure]
- Visual Studio 16 2019
- Win32
- Use default native compiler
[Finish]
```

I built OpenCV-4.5.1 from source, setting the options in CMake-gui:
```
Qt5Gui_DIR		C:/Qt/5.14.1/msvc2017/lib/cmake/Qt5Gui
Qt5_DIR			C:/Qt/5.14.1/msvc2017/lib/cmake/Qt5
BUILD_SHARED_LIBS 		OFF
WITH_DSHOW        		OFF
BUILD_WITH_STATIC_CRT		ON
```

In this case, all /MD (/MDd) keys are automatically changed to /MT (/MTd).
In MSVC, I selected 'Release', then 'Build solution'. I didn't do 'INSTALL'.

I was unable to use the msvc.cmake toolchain, as it gave too many errors, strange flags and a lot of complicated obscure prefixes.
It is a good idea for Sthalik to use the 'sdk-paths-sthalik@MSVC-windows.cmake' file for configuration. I had to rename it to my Windows username 'sdk-paths-Win7@MSVC-windows.cmake'. I think this note about the need for renaming should be written in the Wiki.

At the beginning of the file, I inserted (borrowed) a very useful piece of code from 'OpenCVCRTLinkage.cmake' to replace the /MD (/MDd) keys with /MT (/MTd):
```
  foreach(flag_var
          CMAKE_C_FLAGS CMAKE_C_FLAGS_DEBUG CMAKE_C_FLAGS_RELEASE
          CMAKE_C_FLAGS_MINSIZEREL CMAKE_C_FLAGS_RELWITHDEBINFO
          CMAKE_CXX_FLAGS CMAKE_CXX_FLAGS_DEBUG CMAKE_CXX_FLAGS_RELEASE
          CMAKE_CXX_FLAGS_MINSIZEREL CMAKE_CXX_FLAGS_RELWITHDEBINFO)
    if(${flag_var} MATCHES "/MD")
      string(REGEX REPLACE "/MD" "/MT" ${flag_var} "${${flag_var}}")
    endif()
    if(${flag_var} MATCHES "/MDd")
      string(REGEX REPLACE "/MDd" "/MTd" ${flag_var} "${${flag_var}}")
    endif()
  endforeach(flag_var)
```

In the same file, I set the paths to QT, OpenCV, etc. Those SDKs that I am not ready to build, I disabled using #.
Then everything is as usual in the first:
```
[Configure]:
- Visual Studio 16 2019
- Win32
- Use default native compiler
[Finish]
[Generate]
[Open project] -> Launch C:\Dev\opentrack-build\opentrack.sln (MSVC2019)
```
In MSVC2019, I selected 'Release' then 'Build solution'. Then I selected 'INSTALL' and 'Build INSTALL'.
The size of the 'OpenTrack.exe' file is 406528 bytes. Several times more than with /MD /MDd keys.

Everything worked out, what I could verify works!
Thanks for the help!
Tomorrow I will continue to deal with the rest of the SDKs.

Slion, 2021-02-16:
Kinect SDK is there I believe:
https://www.microsoft.com/en-us/download/details.aspx?id=44561

GO63-samara, 2021-02-17:
@Slion
Which version of the Kinect SDK should I be using?
At first I tried to install the Kinect SDK 2.0, but I got an installer error that I must use Windows8 or older. And I only have Windows7.
Then I installed Kinect SDK 1.8, then Kinect Developer Toolkit v1.8.0 (don't know if I need it?).
Then in 'sdk-paths-Win7@MSVC-windows.cmake' I specified the path set (SDK_KINECT20 "C:/Program Files/Microsoft SDKs/Kinect/v1.8")
When compiling Opentrack in MSVC2019 gave an error:
`tracker-kinect-face\camera_kinect_ir.h (13,10): fatal error C1083: Cannot open include file: 'Kinect.h': No such file or directory`
There is really no such file anywhere.
What am I doing wrong? Should I only be using the Kinect SDK 2.0? Can I somehow install it on Windows7?

Slion, 2021-02-18:
You will need version 2.0, the one for Kinect v2. The older one is for the first Kinect hardware and we don't support it. Strange that it does not work on Windows 7 though.

sthalik, 2021-02-18:
@Slion can you pack and share the include and library directories? Thanks.

Slion, 2021-02-18:
> @Slion can you pack and share the include and library directories? Thanks.

How did we do that back then? Did I package includes and libraries for you? It could be few days until I get around locating those.

sthalik, 2021-02-19:
Alright, in case @GO63-samara has any problems with the SDK installer there's the backup plan.

Slion, 2021-02-20:
I've put it up for download there:
http://publish.slions.net/KinectSDK/Kinect.zip

GO63-samara, 2021-02-21:
Hi @sthalik, Hi @Slion.
Thanks for **Kinect.zip**, I found **Kinect.h** in it and will definitely try using this SDK, but now I had to go back to the OpenCV - Opentrack bundle.

I suddenly noticed that the size of 'opentrack-tracker-pt.dll' is 14393856 bytes, which is almost 10 times larger than in OT-2.3.12.
The rest of the OpenCV-dependent dlls turned out to be just as terribly large. I found this very suspicious.
I took out my old PS3-eye, which I had not included since 2016, installed the driver (CL-Eye-Driver-5.0.1.0218) and tested it in 
PointTracker.
When I click [Camera settings (when available)], the settings window does not open.
When I press [Start], "TRACKING NOT STARTING" disappears and the gray background remains. No video. The red LED on PS3-eye does not turn on.
When tested in the original OT-2.3.12, everything works fine, and even noticeably better than in 2016.
For two evenings I tried to change flags in OT. But the result is zero. I am completely confused and do not know where to dig. I'm new to CMake. It only looks simple, but you can break your head.

In this regard, I have questions:
1. Is such a large dll size due to bugs in my build of OpenCV, OpenTrack or in the linker between them?
2. Should the flags of OpenCV and OpenTrack exactly match?
3. Does PointTracker have any other dependences besides OpenCV? Does he need Eigen or "BOOST_DIR" or something else?
4. When building OpenCV, should I use the "WITH_QT" option and / or "EIGEN3_DIR" and / or the "opentrack-depends" folder?
5. Which version of OpenCV should I be using? I've tried OCV-4.5.1 and OCV-3.4.11. No difference.
6. How to determine if an OpenCV release is built correctly? If the file size "opencv_core3411.lib" is 12159858 bytes  ("opencv_core451.lib" 13342828 bytes) is that too much?

Maybe for a start I take a ready-made x64 release of OpenCV and try with it, like Slion did, then I will understand what the problem is?

Slion, 2021-02-21:
> use CMAKE_TOOLCHAIN_FILE as cmake/msvc.cmake at the time when first configuring. I recommend executng cmake-gui from cmd after enabling vcvars32.bat and selecting Ninja or NMAKE. It should configure itself as MSVC on its own.

I think this is key to having a build output similar to what @sthalik does. Maybe he could just share the actual command lines he uses to make a release usually. Then all you should need to do is make sure you are using the same versions and setup for the various dependencies we have.

sthalik, 2021-02-21:
> Should the flags of OpenCV and OpenTrack exactly match?

Yes. You mentioned build errors with my toolchain file. I'd like to know about them to fix them. All opentrack dependencies (except for Qt) use the toolchain file with my builds.

The build options are hard to understand, which is why I use these resources:

- https://docs.microsoft.com/en-us/cpp/build/reference/compiler-options-listed-by-category?view=msvc-160
- https://docs.microsoft.com/en-us/cpp/build/reference/linker-options?view=msvc-160

> When I click [Camera settings (when available)], the settings window does not open.

Make sure your opencv build is using DSHOW, as per cmake cache variable.

> When building OpenCV, should I use the "WITH_QT" option and / or "EIGEN3_DIR" and / or the "opentrack-depends" folder?

Disable everything from the `WITH_*` group except WITH_DSHOW. Don't build apps or tests in the `BUILD_*` group, it takes forever.

> Which version of OpenCV should I be using? I've tried OCV-4.5.1 and OCV-3.4.11. No difference.

Use master.

sthalik, 2021-02-21:
Also, launch by dragging `opentrack.exe` onto a `cmd.exe` black window, you'll get some debug info.

GO63-samara, 2021-02-22:
> Yes. You mentioned build errors with my toolchain file. I'd like to know about them to fix them. All opentrack dependencies (except for Qt) use the toolchain file with my builds.

Okay, let's sort out the errors in order, with your help.
Today I deleted the contents of the opentrack-build folder and started to deal with msvc.cmake again.
Just in case, to keep things simpler, I went back to the opentrack-2.3.12 release and disabled all SDKs in `sdk-paths-Win7@MSVC-windows.cmake`.
Left only the path `set(Qt5_DIR "C:/Qt/5.14.1/msvc2017/lib/cmake/Qt5" CACHE PATH "" FORCE)`
That is, now no dependencies are used, except for QT.

I built a solution from cmd with the command:
`cmake -S opentrack-2.3.12 -B opentrack-build -DCMAKE_TOOLCHAIN_FILE=cmake/msvc.cmake -DCMAKE_BUILD_TYPE=Release -G "Visual Studio 16 2019" -A Win32`

At first I got the error:
`cl : command line error D8016: '/O2' and '/RTCs' command-line options are incompatible [C:\Dev\opentrack-build\CMakeFiles\CMakeTmp\cmTC_12dc0.vcxproj]
`

To fix the error, I had to add the `-Od` switch, not sure if this is correct:
`set(_CFLAGS_DEBUG "-guard:cf -MTd -Gs0 -RTCs -Od")`

I have flags like this:
```-- -- Compile flags:
--  CXX:
--  CXX_RELEASE: -O2 -O2it -Oy- -Ob3 -fp:fast -GS- -GF -GL -Gw -Gy -arch:SSE2 -MT
-- -- Link flags:
--  LINK_SHARED:
--  LINK_SHARED_RELEASE: -OPT:REF -OPT:ICF=10 -LTCG:INCREMENTAL -DEBUG:FULL
-- -- Static archive flags:
--  STATIC:
--  STATIC_RELEASE: -LTCG:INCREMENTAL
-- --
```
At the end of the main CMakeLists.txt I brought out with the help of many `message` all the other flags:
```
CMAKE_CXX_FLAGS
CMAKE_CXX_FLAGS_DEBUG                   -guard:cf -MTd -Gs0 -RTCs -Od
CMAKE_CXX_FLAGS_MINSIZEREL              /MD /O1 /Ob1 /DNDEBUG
CMAKE_CXX_FLAGS_RELEASE                 -O2 -O2it -Oy- -Ob3 -fp:fast -GS- -GF -GL -Gw -Gy -arch:SSE2 -MT
CMAKE_CXX_FLAGS_RELWITHDEBINFO          /MD /Zi /O2 /Ob1 /DNDEBUG

CMAKE_C_FLAGS
CMAKE_C_FLAGS_DEBUG                   -guard:cf -MTd -Gs0 -RTCs -Od
CMAKE_C_FLAGS_MINSIZEREL              /MD /O1 /Ob1 /DNDEBUG
CMAKE_C_FLAGS_RELEASE                 -O2 -O2it -Oy- -Ob3 -fp:fast -GS- -GF -GL -Gw -Gy -arch:SSE2 -MT
CMAKE_C_FLAGS_RELWITHDEBINFO          /MD /Zi /O2 /Ob1 /DNDEBUG

CMAKE_EXE_LINKER_FLAGS
CMAKE_EXE_LINKER_FLAGS_DEBUG            -DEBUG:FULL
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL       /INCREMENTAL:NO
CMAKE_EXE_LINKER_FLAGS_RELEASE          -OPT:REF -OPT:ICF=10 -LTCG:INCREMENTAL -DEBUG:FULL
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO   /debug /INCREMENTAL

CMAKE_MODULE_LINKER_FLAGS
CMAKE_MODULE_LINKER_FLAGS_DEBUG          -DEBUG:FULL
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL     /INCREMENTAL:NO
CMAKE_MODULE_LINKER_FLAGS_RELEASE        -OPT:REF -OPT:ICF=10 -LTCG:INCREMENTAL -DEBUG:FULL
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO /debug /INCREMENTAL

CMAKE_RC_FLAGS                         /nologo /DWIN32
CMAKE_RC_FLAGS_DEBUG                   -D_DEBUG
CMAKE_RC_FLAGS_MINSIZEREL
CMAKE_RC_FLAGS_RELEASE
CMAKE_RC_FLAGS_RELWITHDEBINFO

CMAKE_SHARED_LINKER_FLAGS
CMAKE_SHARED_LINKER_FLAGS_DEBUG          -DEBUG:FULL
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL     /INCREMENTAL:NO
CMAKE_SHARED_LINKER_FLAGS_RELEASE        -OPT:REF -OPT:ICF=10 -LTCG:INCREMENTAL -DEBUG:FULL
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO /debug /INCREMENTAL

CMAKE_STATIC_LINKER_FLAGS
CMAKE_STATIC_LINKER_FLAGS_DEBUG
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL
CMAKE_STATIC_LINKER_FLAGS_RELEASE          -LTCG:INCREMENTAL
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO
```
Then I ran 'Opentrack.sln' in MSVC2019 and selected [Release] then [Build solution].
I got 23 Errors, 23 Warnings.

They are all something like this:
```
Warning	LNK4075	ignoring '/INCREMENTAL' due to '/LTCG' specification	opentrack-api	C:\Dev\opentrack-build\api\LINK	1	
Error	LNK1218	warning treated as error; no output file generated	opentrack-api	C:\Dev\opentrack-build\api\LINK	1	

```
Interestingly, if I select [RelWithDebInfo] everything works and I can start Opentrack. Some kind of confusion with flags in msvc.cmake, is not it?
How to fix it?

sthalik, 2021-02-22:
> Interestingly, if I select [RelWithDebInfo] everything works and I can start Opentrack. Some kind of confusion with flags in msvc.cmake, is not it?

Use the Release profile. The toolchain file doesn't support multi-profile generators. For normal generators it already forces the release build type.

Normally I'd do it like this. Open a console window, drag in `vcvars32.bat`, run cmake in a `build` subdirectory:

`cmake -DCMAKE_TOOLCHAIN_FILE=%CD%/../cmake/msvc.cmake -GNinja ..`

Using the `.sln` generator is asking for trouble long-term.

GO63-samara, 2021-02-23:
Wow! It works!

The `ninja install` command helped me. I tried to use `-G Ninja` before, but I just didn't know what to do with the `build.ninja` file.

I used this commands from `cmd`:
```
C:\MSVS\2019\Community\VC\Auxiliary\Build\vcvars32.bat
cmake -S opentrack-2.3.12 -B opentrack-build -DCMAKE_TOOLCHAIN_FILE=cmake/msvc.cmake -G Ninja
cd opentrack-build 
ninja install
```
As a result, I got the INSTALL folder with working Opentrack, no errors. All dll sizes are close to those in OT-2.3.12. Many files have the same size.

Thank You!

sthalik, 2021-02-23:
Good to hear that. How many modules can you get to work? Is the Wii tracker working?

GO63-samara, 2021-02-23:
I am now just learning how to compile a release from the OT-2.3.12 sources in order to exclude possible errors in the opentrack-master.
I just managed to build a release of OpenCV and connect it to OpenTrack. OpenCV works on builds from `sln` (without mcvs.cmake) and `ninja` (with msvc.cmake).

PointTracker runs and works well in both OpenCV builds. But in ninja build, the dll size is smaller.

[image]

I managed to check the work of trackers:

- PointTracker
- Hatire Arduino
- UDP over network

I do not have WII, but it is in the list of trackers, and when I run it I get a picture:

[image]

Most likely it works too.

It looks like EasyTracker is not working. Opentrack closes with an error when EasyTracker launched:
```
Сигнатура проблемы:
  Имя события проблемы:	APPCRASH
  Имя приложения:	opentrack.exe
  Версия приложения:	0.0.0.0
  Отметка времени приложения:	60353e47
  Имя модуля с ошибкой:	opentrack-tracker-easy.dll
  Версия модуля с ошибкой:	0.0.0.0
  Отметка времени модуля с ошибкой:	60353e93
  Код исключения:	40000015
  Смещение исключения:	00173f96
  Версия ОС:	6.1.7601.2.1.0.256.1
  Код языка:	1049
  Дополнительные сведения 1:	30e7
  Дополнительные сведения 2:	30e76cd94bc5eeef84a5f7e09dc885e2
  Дополнительные сведения 3:	9b86
  Дополнительные сведения 4:	9b867d797d736c5bea3725a3016001ce

Ознакомьтесь с заявлением о конфиденциальности в Интернете:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0419

Если заявление о конфиденциальности в Интернете недоступно, ознакомьтесь с его локальным вариантом:
  C:\Windows\system32\ru-RU\erofflps.txt
```
The modules folder now contains the following files:
```
30.06.2020  23:35                97 cleye.config
30.06.2020  23:35            73,216 freetrackclient.dll
30.06.2020  23:35            16,384 freetrackclient64.dll
30.06.2020  23:35            15,872 NPClient.dll
30.06.2020  23:35            19,968 NPClient64.dll
23.02.2021  21:40           120,320 opentrack-filter-accela.dll
23.02.2021  21:40           138,240 opentrack-filter-ewma.dll
23.02.2021  21:40           238,080 opentrack-filter-hamilton.dll
23.02.2021  21:41           178,688 opentrack-filter-kalman.dll
23.02.2021  21:40           116,736 opentrack-proto-fgfs.dll
23.02.2021  21:40           154,112 opentrack-proto-freetrack.dll
23.02.2021  21:40           101,888 opentrack-proto-fsuipc.dll
23.02.2021  21:40           118,272 opentrack-proto-simconnect.dll
23.02.2021  21:40           114,176 opentrack-proto-udp.dll
23.02.2021  21:40           111,104 opentrack-proto-vjoy.dll
23.02.2021  21:40           264,192 opentrack-proto-win32-mouse.dll
23.02.2021  21:42         2,394,624 opentrack-tracker-easy.dll
23.02.2021  21:40           128,000 opentrack-tracker-freepie-udp.dll
23.02.2021  21:40           168,448 opentrack-tracker-fusion.dll
23.02.2021  21:40           367,104 opentrack-tracker-hatire.dll
23.02.2021  21:40           108,544 opentrack-tracker-hydra.dll
23.02.2021  21:40           132,096 opentrack-tracker-joystick.dll
23.02.2021  21:42         1,813,504 opentrack-tracker-pt.dll
23.02.2021  21:40           134,656 opentrack-tracker-s2bot.dll
23.02.2021  21:40           102,400 opentrack-tracker-test.dll
23.02.2021  21:40           117,760 opentrack-tracker-udp.dll
23.02.2021  21:42         1,851,904 opentrack-tracker-wii.dll
23.02.2021  21:41         1,435,648 opentrack-video-opencv.dll
27.10.2019  01:56           622,080 sixense.dll
30.06.2020  23:35           386,048 TrackIR.exe
27.10.2019  01:56           141,824 vJoyInterface.dll

```
Are there any additional dependencies for EasyTracker?

sthalik, 2021-02-24:
> It looks like EasyTracker is not working. Opentrack closes with an error when EasyTracker launched

OpenCV assertions show on stdout so launch opentrack from a console window. Also the toolchain generates `.pdb` files in `install/debug` so you can get a backtrace by attaching a debugger after the crash. Use ctrl+alt+P in Visual Studio.

sthalik, 2021-02-24:
For the version to show in the title bar, make sure `git.exe` is in PATH. Normally the Git install directory has a "cmd" subdirectory with the wrapper executable. Then navigate to the build directory, run `cmake .` and then `ninja install` again.

The Wii tracker is fine. Previously it wasn't shown at all despite linking correctly.

Slion, 2021-02-24:
> Are there any additional dependencies for EasyTracker?

Indirect dependency with Kinect V2 I think. It's expecting an IR 8bits feed from the camera if I recall well. So it could be that it just crashes with normal color camera feeds. If it built it's probably working just fine when configured to use Kinect's IR camera.

sthalik, 2021-02-25:
@GO63-samara I forgot, for EasyTracker you need HAVE_WIN32UI in OpenCV in addition to DSHOW.

GO63-samara, 2021-02-25:
Thanks for the help.

First about EasyTracker. It seems to work.

> OpenCV assertions show on stdout so launch opentrack from a console window. Also the toolchain generates .pdb files in install / debug so you can get a backtrace by attaching a debugger after the crash. Use ctrl + alt + P in Visual Studio.

> I forgot, for EasyTracker you need HAVE_WIN32UI in OpenCV in addition to DSHOW.

I do not have an `opentrack-build/install/debug` folder, but there are many `*.pdb` files and they are scattered in different folders in `opentrack-build`.
In the begin, when running from `cmd` and starting EasyTracker, opentrack wrote this:
```
16142018352086492: Update model - begin
16142018352106493: Update model - end
16142018352126494: Update Setting - begin
16142018352126494: Update Setting - end[C:\OpenCV\opencv-4.5.1\modules\highgui\src\windows.cpp:634] opencv: The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script at cvNamedWindow
```
I enabled `WITH_WIN32UI ON` and EasyTracker now starts.
OpenCV currently has these settings:
```
BUILD_SHARED_LIBS OFF
BUILD_WITH_STATIC_CRT ON
WITH_DSHOW ON
WITH_WIN32UI ON
Others WITH_* OFF

```
EasyTracker displays video at startup in the main window and in the Preview window:

[image]

[image]

The octopus doesn't move. Should I deal with the settings?

GO63-samara, 2021-02-25:
Now about Kinect. It looks like it worked with SDK from @Slion.
First when running from `cmd`, opentrack wrote this:
```
C:\Dev\opentrack-build\install>
DEBUG [C:\Dev\opentrack-2.3.12\api/plugin-support.hpp:107]: library "kinect-face
" failed: "Cannot load library C:\\Dev\\opentrack-build\\install\\modules\\\\ope
ntrack-tracker-kinect-face.dll:
```
I thought the dll doesn't work under Windows7. Then I downloaded and installed Windows 10 on another HDD. I haven't even configured it yet without any drivers. But in Windows 10, when launching OpenTrack, there was exactly the same message.

When launching OpenTrack in Windows 10, I first got an error that the `msvcp140.dll` and `vcruntime140.dll` libraries were not found. I just copied them to Opentrack from MSVC2019. Not sure if I did the right thing?

Then I guessed to install `KinectRuntime-v2.0_1409-Setup.exe` and the `opentrack-tracker-kinect-face.dll` error disappeared, Kinect appeared in the list of trackers and even began to run, only without video (does it works better with the camera driver?).

[image]

The settings window is as follows:

[image]

When trying to install `KinectRuntime-v2.0_1409-Setup.exe` in Windows7, I got this message:

[image]

I am wondering if I have an HDD with KinectRuntime-v2.0 installed under Windows 10, can I somehow copy it to the HDD with Windows 7?

GO63-samara, 2021-02-25:
Also I installed Aruco. I just had to build it from source first, using `vcvars32.bat`, `cmake-gui`, `Ninja` and `msvc.cmake`.
The video shows. The octopus doesn't move. I don't have an Aruco marker.

> For the version to show in the title bar, make sure git.exe is in PATH. Normally the Git install directory has a "cmd" subdirectory with the wrapper executable. Then navigate to the build directory, run cmake . and then ninja install again.

Ok, I'll use this later.

But I still need to deal with SDK_REALSENSE, SDK_RIFT_140 and SDK_VALVE_STEAMVR
And I'm not clear what the SDK_PS3EYEDRIVER is for? SDK_PS3EYEDRIVER is enabled for me, but no files appear in modules.

sthalik, 2021-02-25:
> I thought the dll doesn't work under Windows7. Then I downloaded and installed Windows 10 on another HDD. I haven't even configured it yet without any drivers. But in Windows 10, when launching OpenTrack, there was exactly the same message.

The Kinect library fails to load unless you have the runtime actually installed. When in doubt, use Dependency Walker. That program's been invaluable for opentrack development over the years.

> When launching OpenTrack in Windows 10, I first got an error that the msvcp140.dll and vcruntime140.dll libraries were not found. I just copied them to Opentrack from MSVC2019. Not sure if I did the right thing?

Technically, the Kinect runtime will install the required CRT files. But you can't go wrong copying them into program's install directory. One thing to avoid is ever requiring the user to run these redistributable installers, which are slow and annoying.

> SDK_VALVE_STEAMVR

https://github.com/ValveSoftware/openvr

Use the binary release.

> SDK_RIFT_140

https://developer.oculus.com/downloads/native-windows/

> And I'm not clear what the SDK_PS3EYEDRIVER is for? SDK_PS3EYEDRIVER is enabled for me, but no files appear in modules.

It's the open PS3 Eye driver so that people don't need to pirate the paid driver anymore. You need libusb for that.

> SDK_REALSENSE

https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html

You need the essentials and "face tracking and recognition" modules. Alas, registration is required.

GO63-samara, 2021-02-28:
Only now I was able to build trackers `Valve SteamVR`, `Oculus Rift runtime 1.4.0 - HMD`, `Intel RealSence Technology`.
There are now 16 trackers on the list in Win10. In Win7, one less tracker, `Kinect` is not displayed.
[image]

> SDK_VALVE_STEAMVR
> https://github.com/ValveSoftware/openvr
> Use the binary release.

Ninja install gives an error with this OpenVR SDK version 1.16.8:
```
SDK_VALVE_STEAMVR
https://github.com/ValveSoftware/openvr
Use the binary release."
Ninja install даёт ошибку с этой версией OpenVR SDK 1.16.8:
"
C:\Dev\opentrack-2.3.12\tracker-steamvr\steamvr.cpp(257,1): error C2039: 'ResetS
eatedZeroPose': is not a member of 'vr::IVRSystem'
                v->ResetSeatedZeroPose();
^
C:\Dev\opentrack-depends\openvr-1.16.8\headers\openvr.h(2066): note: see declara
tion of 'vr::IVRSystem'
class IVRSystem
C:\Dev\opentrack-2.3.12\tracker-steamvr\steamvr.cpp(257,1): error C2039: '__this
': is not a member of 'vr::IVRSystem'
                v->ResetSeatedZeroPose();
^
C:\Dev\opentrack-depends\openvr-1.16.8\headers\openvr.h(2066): note: see declara
tion of 'vr::IVRSystem'
class IVRSystem

```
Valve SteamVR now works with OpenVR SDK 1.3.22: https://github.com/opentrack/openvr
`setq(SDK_VALVE_STEAMVR "openvr-1.3.22")`
After clicking [Start], a window opens with an error:
[image]

> SDK_RIFT_140
> https://developer.oculus.com/downloads/native-windows/

Here Oculus Rift runtime 1.4.0 - HMD started working without problems with the version:
`setq(SDK_RIFT_140 "ovr_sdk_win_23.0.0/LibOVR")`
After clicking [Start], a window opens with an error:
[image]

```
SDK_REALSENSE
https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html
You need the essentials and "face tracking and recognition" modules. Alas, registration is required.
```
After installing the SDK (intel_rs_sdk_mw_core_offline_11.0.27.1384.exe, intel_rs_sdk_mw_face_hybrid_11.0.28.2787.exe) and setting `set (SDK_REALSENSE "C: / Program Files (x86) / Intel / RSSDK")` tracker `Intel RealSence Technology` appeared
After starting a window appears:

[image]

After clicking [OK] Opentrack closes with an error:

```
Сигнатура проблемы:
  Имя события проблемы:	BEX
  Имя приложения:	opentrack.exe
  Версия приложения:	0.0.0.0
  Отметка времени приложения:	603bc978
  Имя модуля с ошибкой:	libpxcface_idc.dll_unloaded
  Версия модуля с ошибкой:	0.0.0.0
  Отметка времени модуля с ошибкой:	5852fdaa
  Смещение исключения:	6beebb19
  Код исключения:	c0000005
  Данные исключения:	00000008
  Версия ОС:	6.1.7601.2.1.0.256.1
  Код языка:	1049
  Дополнительные сведения 1:	0a9e
  Дополнительные сведения 2:	0a9e372d3b4ad19135b953a78882e789
  Дополнительные сведения 3:	0a9e
  Дополнительные сведения 4:	0a9e372d3b4ad19135b953a78882e789

```
Some problems with RealSense.

sthalik, 2021-03-01:
> SDK_VALVE_STEAMVR 

Try replacing `ResetSeatedZeroPose` with `ResetZeroPose(vr::TrackingUniverseSeated)`.

> SDK_RIFT_140

This is fine, you don't have the device or the drivers.

> SDK_REALSENSE

This is all fine, except for the crash that is.

GO63-samara, 2021-03-01:
> Try replacing ResetSeatedZeroPose with ResetZeroPose(vr::TrackingUniverseSeated).

I tried, then I get the following error:
```
C:\Dev\opentrack-2.3.12\tracker-steamvr\steamvr.cpp(257,1): error C2039: 'ResetZ
eroPose': is not a member of 'vr::IVRSystem'
                v->ResetZeroPose(vr::TrackingUniverseSeated);
^
C:\Dev\opentrack-depends\openvr-1.16.8\headers\openvr.h(2066): note: see declara
tion of 'vr::IVRSystem'
class IVRSystem
C:\Dev\opentrack-2.3.12\tracker-steamvr\steamvr.cpp(258,1): error C2039: '__this
': is not a member of 'vr::IVRSystem'
                v->ResetZeroPose(vr::TrackingUniverseSeated);
^
C:\Dev\opentrack-depends\openvr-1.16.8\headers\openvr.h(2066): note: see declara
tion of 'vr::IVRSystem'
class IVRSystem
```
Apparently the IVRSystem class has changed.
In my opinion, there is no need to work with OpenVR SDK 1.16.8 now. It is enough that OpenTrack works with openvr-1.3.22.
You can improve it in the next versions of OpenTrack.

GO63-samara, 2021-03-01:
Compared to the original Opentrack-2.3.12 my `install/modules` folder is missing files:
```
libusb-1.0.dll
opentrack-video-ps3eye.dll
ps3eye-frame-test.exe
ps3eye-mode-test.exe
ps3eye-subprocess.exe
```
My guess is that SDK_LIBUSB and SDK_PS3EYEDRIVER need to be compiled.
I am trying to use `opentrack-depends/libusb-msvc-x86` and https://github.com/opentrack/PS3EYEDriver.
I have set:
```
setq(SDK_LIBUSB "libusb-msvc-x86")
setq(SDK_PS3EYEDRIVER "PS3EYEDriver")
```
But SDK_PS3EYEDRIVER is not used anywhere in OpenTrack and CMakeLists.txt from `opentrack-depends/PS3EYEDriver` folder is not even executed.
How should I install SDK_LIBUSB and SDK_PS3EYEDRIVER correctly for OpenTrack?

sthalik, 2021-03-01:
> But SDK_PS3EYEDRIVER is not used anywhere in OpenTrack and CMakeLists.txt from opentrack-depends/PS3EYEDriver folder is not even executed.

Run `git submodule update --init`.

GO63-samara, 2021-03-02:
> Run git submodule update --init.

Everything worked out! Thank you!
Before that I did via download https://github.com/opentrack/opentrack/archive/opentrack-2.3.12.zip
It turns out that it was necessary to do it through cloning:
```
cd C:\Dev
git clone https://github.com/opentrack/opentrack.git
git checkout 0f13f02f2fdce44934476665a9e3bc97c1fccb82

HEAD is now at 0f13f02f video/ps3eye: update submodule
M       video-ps3eye/ps3eye-driver

git submodule update --init

Submodule path 'video-ps3eye/ps3eye-driver': checked out '2cde6fbfbde7874daab634
ca3b35b7ede14ed88c'
```
The `install/modules` folder now contains all the files. `PS3 open driver` appeared and works:

[image]

Suddenly, the correct caption appeared at the Opentrack window:

[image]

Questions remain:
- There is a `setq(SDL2_DIR "SDL2-win32")` that does nothing. I don't have this library. What is this library and where can I get it if needed?
- My `install` folder is missing the `debug` folder with 40 `*.pdb` files. How critical is this, and how to build this folder correctly, if needed?
- I noticed that when Opentrack crashes after launching `Intel RealSense` tracker, then `TrackIR.exe *32` process remains hanging. So RealSense tracker isn't working? But in the original Opentrack-2.3.12, everything is the same.

sthalik, 2021-03-03:
Set `opentrack_install-debug-info` to true.

> There is a setq(SDL2_DIR "SDL2-win32") that does nothing. I don't have this library. What is this library and where can I get it if needed?

This is for debugging the driver. I used to install an executable showing the camera feed.

> So RealSense tracker isn't working? But in the original Opentrack-2.3.12, everything is the same.

What do you mean by "the same", does it work or not?

GO63-samara, 2021-03-03:
> Set opentrack_install-debug-info to true.

Thanks! I succeeded, now I have a `install/debug` folder with 40 `*.pdb` files. But should it be in release? It weighs 333MB.

> This is for debugging the driver. I used to install an executable showing the camera feed.

Hmm. I think that such a utility would be useful for checking the camera driver, as part of Opentrack or separately.

> What do you mean by "the same", does it work or not?

I don’t have a Realsense camera, I can’t check for sure. To clarify, when launching Realsense tracker under Win10, an error window appears and Opentrack continues to work normally.
When launching Realsense tracker under Win7, a window appears with an error and Opentrack crashes, writes this in the console:
```
DEBUG [C:\Dev\opentrack\tracker-rs\ftnoir_tracker_rs.cpp:64]: tracker_rs: starti
ng preview
DEBUG [C:\Dev\opentrack\tracker-rs\ftnoir_tracker_rs_worker.cpp:25]: tracker_rs:
 tracking has started. retValue:  -3
DEBUG [C:\Dev\opentrack\tracker-rs\ftnoir_tracker_rs.cpp:77]: tracker_rs: stoppe
d preview
DEBUG [C:\Dev\opentrack\logic\pipeline.cpp:562]: tracker: backlog interval overf
low 15492.1 ms
```
The `TrackIR.exe *32` (3576 KB) process remains hanging in the Task Manager. I close it manually in the manager.
This behavior is the same in both the Opentrack I built and the original Opentrack-2.3.12.
Apparently, Realsense tracker does not work under Win7, but only under Win8 and Win10.

sthalik, 2021-03-03:
> Thanks! I succeeded, now I have a install/debug folder with 40 *.pdb files. But should it be in release? It weighs 333MB.

The .pdb files aren't deterministic so they must be distributed with the app. Use 7-zip with all options on max to get it to normal size.

> Hmm. I think that such a utility would be useful for checking the camera driver, as part of Opentrack or separately.

There are a few frame-test executables for various parts of the driver. Only the GUI visualizer depends on SDL so I don't ship it. It should build and work though.

> When launching Realsense tracker under Win7, a window appears with an error and Opentrack crashes

Press CTRL+ALT+P in Visual Studio to attach, then make it crash. There are 2 threads, GU+`logic/pipeline.cpp` and QThread from the module.

> The TrackIR.exe *32 (3576 KB) process remains hanging in the Task Manager. I close it manually in the manager.

That only happens on crashes. I don't close the process by name, only by handle so crashes during development are a problem. Also, disable Fault Tolerant Heap in Windows 10 `gpedit.msc` to save yourself the pain.

> Apparently, Realsense tracker does not work under Win7, but only under Win8 and Win10.

Does this mean that it only crashes on win7?

GO63-samara, 2021-03-04:
> Press CTRL+ALT+P in Visual Studio to attach, then make it crash. There are 2 threads, GU+logic/pipeline.cpp and QThread from the module.

Well, I created a new empty project in MSVC, then I launched the OpenTtrack, in MSVC pressed `CTRL + ALT + P` and made an attach to the Opentrack process, then I crash Opentrack, and then I don't know what to do in MSVC.
Maybe this is superfluous? In Windows 10, everything works as it should.

> Does this mean that it only crashes on win7?

Yes, Opentrack only crashes on Windows 7. On Windows 10, the error window closes with the [OK] button normally, without Opentrack crashing.

sthalik, 2021-03-04:
> Maybe this is superfluous? In Windows 10, everything works as it should.

I think so too. I'd say just ignore it unless you specifically want to pursue this issue.

You should already have read/write repo and release rights. The question is whether you want to write the release notes yourself.

GO63-samara, 2021-03-05:
Today I rebuilt the release already from the `master` branch https://github.com/opentrack/opentrack
I started from the very beginning, with cloning and update to repeat everything. Everything worked out, Opentrack works. I got a strange caption again, is that okay?

[image]

- Pose-widget works in a new way.
- Tobii is missing in the tracker list and in the `install/modules` folder. Should I download some `SDK_TOBII`?
- The red LED on the PS3 still remains on when pressing [Stop] with the PS3 eye open driver. With PS3Eye camera driver, the red LED turns on and off correctly.
- Not sure how to enable dark theme to check.

In my opinion, need to reduce the size of the pose-widget so that it is the same size as the video. The Opentrack window will also shrink to take up less screen space.

> You should already have read/write repo and release rights. The question is whether you want to write the release notes yourself.

Are you suggesting that I publish the release myself? I have never done this yet. I would prefer to send you an archive with an `install` folder so you can publish it correctly.

sthalik, 2021-03-05:
> I got a strange caption again, is that okay?

Make a tag for next 2.3.x, `cmake .` in the build dir and rebuild. Check if it works and then push the tag.

> Are you suggesting that I publish the release myself? I have never done this yet. I would prefer to send you an archive with an install folder so you can publish it correctly.

Given the work you've done on getting it to build, I think it's better if you get to actually release it. Unless you really don't want to do that for any reason.

> Tobii is missing in the tracker list and in the install/modules folder. Should I download some SDK_TOBII?

Tobii hasn't been given enough review or work so it's best to leave it as-is.

> The red LED on the PS3 still remains on when pressing [Stop] with the PS3 eye open driver. With PS3Eye camera driver, the red LED turns on and off correctly.

That's not critical, even if it may throw some people off. I've tried to fix it but the fix didn't work. It's enough for now that it provides the video feed.

> Not sure how to enable dark theme to check.

See: https://github.com/opentrack/opentrack/commit/32c8c82b963e349a8345a24c7a5e3597632c2f9c#diff-737207d7a700db2f57633b594bf6c2d5ef58c9b5558c420e0b43e622a5e9129fR165

GO63-samara, 2021-03-08:
> See: 32c8c82#diff-737207d7a700db2f57633b594bf6c2d5ef58c9b5558c420e0b43e622a5e9129fR165

Thanks, I checked out the dark theme. It only works when the dark theme is enabled in Windows 10:

[image]

I saw a mismatch in the colors and sizes of the widgets and fixed them in the Pull request: https://github.com/opentrack/opentrack/pull/1240
Now the video and octopus widgets are the same size, and the size of the main window has been reduced to the same size as in Opentrack-2.3.10. The light background in the pose-widget no longer interferes with both light and dark themes.

[image]

[image]

How about merge it to master and do release with these changes already?

sthalik, 2021-03-08:
> How about merge it to master and do release with these changes already?

Looks good for release. Great work.

GO63-samara, 2021-03-09:
Can you merge https://github.com/opentrack/opentrack/pull/1240/commits/13e9b3ab23da11e8092dbed9a1e57d35b50ebc73 to `master` ?
Is it time to do `git tag opentrack-2.3.13`?

sthalik, 2021-03-09:
I just did. Normally you can do the same, using the green merge button on the bottom left.

It's time. 🚀

GO63-samara, 2021-03-10:
> I just did. Normally you can do the same, using the green merge button on the bottom left.

No, I couldn't. The [Merge] button was inactive. And there was a message:
`Only those with write access to this repository can merge pull requests.`

> It's time. rocket

I created a tag:
`c:\Dev\opentrack>git tag opentrack-2.3.13`
And rebuilt Opentrack to check. The header is now correct: 

[image]

Verification:
```
c:\Dev\opentrack>git show
commit 862dd8c03128935f1dcd52a4eda0b43999e2aa97 (HEAD -> master, tag: opentrack-
2.3.13, origin/master, origin/HEAD)
Merge: 058942f4 13e9b3ab
Author: Stanislaw Halik <sthalik@misaki.pl>
Date:   Tue Mar 9 22:29:54 2021 +0000
    Merge pull request #1240 from GO63-samara/master
    Fixed the sizes of the widgets in the main window
```
I try to push and get rejected:
```
c:\Dev\opentrack>git push origin --tags
remote: Permission to opentrack/opentrack.git denied to GO63-samara.
fatal: unable to access 'https://github.com/opentrack/opentrack/': The requested
 URL returned error: 403
```
I tried both through `login / password` and through `Personal Access Token`, in which I put checkboxes on Repo. I tried on both Windows 7 and Windows 10. Even installed the latest `Git-2.30.2`. Nothing turns out.
What am I doing wrong? I don’t have write access rights?

sthalik, 2021-03-11:
Sorry, I added you to the project but forgot to give you write access. It should work now.

GO63-samara, 2021-03-11:
Thank! Now I managed to do `git push origin --tags`
And I see that there are `zip` and `tar.gz` archives in `opentrack/tags`
What should I do now to submit the install folder?
Should I click `[ Draft a new release ]`?
The release should have a Notes list. Will you write it?

sthalik, 2021-03-13:
Draft release, select tag. Upload the portable version (7z ultra quality, solid archive) and the installer. Write or leave the notes empty, whichever you prefer. 👍

GO63-samara, 2021-03-13:
Yes, I did it! March 13 - version 13.
I can't format the notes text. How to do it?
@Slion Can you test Easy Tracker and Kinect in this release?

[10 more replies]

closed 2022-09-30 by Slion