opentrack/opentrack #1229: Point tracker help please.

Iknowlongnamerightwhatwasithinking, 2021-01-29:
I have been away for awhile and have come back to try to sort his out. 

I have a custom 3p  "model" I want to use it with the pointracker  tracker. 

I cannot find a simple explanation about how to set up the point tracker, how the dimensions work etc. 
The point tracker web page really does not help. 

Issues like: If I look round too much it starts to look up or down. If I look up too much looking round is very sensitive. 

I have tried the "calibrate" tool but that is putting the yellow cross ( what ever that is ) way way off in some random place. 

Can we please get some light shed on this "dark art" it's just nuts that there is no single document that adequately explains how 
set up this point tracker thingy.  If someone could explain the details I would write up a document. ... 

Can we PLEASE put this to rest. ( and just because it works for you does not mean it works for all ) 

This has been a problem for years that I have tried to ignore by making weird arse responses to try to mitigate things. 
I am pretty sure this was never a problem in Freetrack ( I loved that program... ) 

HELP !!! 

Thanks! : - ) 

Robert.

MathijsG, 2021-02-16:
What dimensions does your clip have? And why did you choose to use a different dimension?

Calibration sets the yellow cross on the place where the neck joint is located.

Iknowlongnamerightwhatwasithinking, 2021-02-19:
Hello M.  I don't use a "clip". I use the hat 3p IR target. Without a hat. Yes, about those dimensions.... no idea. I have experimented with changing settings in the calibrate page and well things just went all over the place. I have tried the auto cal and that put my "neck joint" way way off to one side. ( ??? ) I am assuming that my target being beyond redemption.  Frankly I wish there was a HOW TO. On designing a custom target. I am not a hat fan.  AND I don't need all the axis. Slide X,Slide Y,Rot X Rot Y would be fine..... I suppose I could in theory make a hat and try that out. OR just go back to using 1p. ( Single point tracker ... it's pretty darn robust. ) 
Actually starting from scratch with a two axis tracker that is designed to give the best resolution. I find that the 6axis 3p set up uses the LEAST sensor real estate.  It's just crazy.  The Rot X/Y being the most used axis gets the least sensor pix .....  but anyway this is neither here not there as no one in their right mind would do that.  Hardcore ACM with the best Rot X/Y is where it is at. OR this: You can toggle between super resolution 2 axis and lower res 6 axis. In the heat of the engagement you are probably Rot x/y the most and sliding x/y/z the least, or not at all.  Checking my six, looking up, scanning round... They come first. Looking over the side, round struts etc checking gauges etc come second. Switching between the two would be cool.

MathijsG, 2021-02-22:
You can just leave all the axes that you don't want to zero in the graphs, so they don't do anything.

I don't really understand what not to understand about the dimensions. Just measure your dimensions between the shown leds:

[image]

The 60 MM is the height difference between the lower front leds and the higher led in the back. 

The 100mm is the length of the back led in relation to the front led.

The 40mm is the width between one of the outer leds and the middle led.

I use this cap model and it works perfectly: https://www.thingiverse.com/thing:707734

Make sure that during the calibration the camera never loses track of the leds. So don't look too far away from the screen, just make some tiny horizontal and vertical motions. The yellow cross should be at the neck joint's position.

Iknowlongnamerightwhatwasithinking, 2021-02-22:
sure you input the model dimensions. Check. 
But the model to head relation?  

If I use the auto calib it moves the + miles away from my  actual head pivot point. 
( Yes I can seem my actual head and neck in the video, so I can tell...) 

I don't wear a cap. I made a headband and then put the 3p cap target on that. 
Jeez  labeling the LEDS  A B C would be a bloody good start....  from the cameras POV 
left to right .  Left A middle B right C. ok? 

So B is over my head. If I rotate it does not move. Check? 
The A to C is about 120 mm. and the AC to B is about 100. 
I have tried the auto cal and it goes ape.  I have tried changing the  fields. nix. 
I normally leave the yellow + at the B point.  

So I have to wear a cap and make a target that fits into that config ?  I am not a cap fan. 

The A and C leds do they have to stick waay out the front for this work?   How much leeway is there?  I guess if Rot X and Y has lots of SWING and I reduce this then things go out of range ? 
( the swing is A and C rotating as I move my head for X and Y. )  

I experimented with moving Camera and static Target. 1p tracker. see here:
https://guuguu123456789.web.fc2.com/
Use old ver of OT. It was interesting. Way better use of rot X and Y pixles.  but later Ver OT would not support. 

I loved FT.... moved to OT as FT was bugging out.  Shame. 

Got to run.  

S!

MathijsG, 2021-02-22:
The model to head relation is all made via the axes mappings.

Maybe you should film the problem, because it's quite hard to follow. What exactly goes "ape"?

Some images or video's would be much more helpful in understanding the problem.

Or what if you use the Trackhat version: https://www.trackhat.org/trackhat-opentrack

Freetrack was buggy as hell here, on everything system newer than Windows XP.

Cap or no cap of course doesn't matter at all, as the software doesn't even know something about a cap. As long as you have made a correct model. Any pictures of your tracker?

Iknowlongnamerightwhatwasithinking, 2021-02-23:
If I use auto cal it puts the Yellow + miles away from my neck head pivot point. 
This is what I mean by going ape. It's no where near. Miles away. 
If I move the + by entering values in the field it just does weird things in game. 
 Like: if I look up too much I rotate, or if I look down it rotates. 

I am away from PC  and out of town so no pics vids etc.   

I am now assuming that the cap 
tracker has to actually be tracking a very close to "normal" cap model.  What are the normal dimensions? 
If you were to make a cap tracker model without the cap.  Say a headband. 
A decent diagram with the users HEAD and the CAP and the CAMERA to explain the actual details.  Even how you wear it and how you sit in relation to the camera must be a deal. Right? 
Too close, too far away, cap tilted up or down, sitting too low too high etc. 

and no, I am not going near that trackhat nonsense.  Talk about ganking someone else's hard work. Jeebers. 

I still love FT. If it ran I would have kept using it. I used it for ages on Win 7 but trying to get it to start was a grind. I finally gave up. SMH.  I ganged Rot X to Slide X in FT so I could rot and slide as one. It was pretty darn good for a simple work around for 1p. 

And no, simply unassigned Slide X and Y does not help that much. Maybe if A and C leds were further out front it might.  The further out those leds are the more they move over the FOV. 
This would increase the amount of pix from straight ahead to fully left or right. 

If there was a model for only 4 axis and it was designed for super resolution: it uses as much of the sensor as possible with Rot X Y getting the most pix. and slide X Y getting the least.  I hardly used Slide Y.  Slide X is good for looking over the side of AC, or looking past the rudder etc. 

I am going to try 1p with 1p tracker and old version of OT.  just for kicks.  

What happens if you have a massive screen? The camera is going to be so offset....  
Lucky I only have this old small one. I guess.

Ok I give up for this moment. I will be back at pc in a few days. 

S!