opentrack/opentrack #1231: Jerky movements with PS3 Eye Camera and delanclip

iAmEMT, 2021-02-10:
I got a delanclip in December and used it for a few weeks with no issues, and just yesterday I loaded up DCS to play again and now all of my head movements are stuttery in game and the octopus shows the jerking as well. I suspect it may be the camera FPS drops I see in the settings, but has anyone else experienced this? I've already reinstalled the CL-Eye Test driver and referenced https://github.com/opentrack/opentrack/issues/1182 with no success. I have not been able to test facetracknoir yet.

https://user-images.githubusercontent.com/23224672/107480090-a928c600-6b41-11eb-8cb4-0193833718fc.mp4

sjlaurin, 2021-02-10:
Same as you, my Opentrack+Declanclip was running fine since November 2020. I fired up my game yesterday and now I have a lot of jitter. I've been using the same hardware and profile for a while now. 
 
- My Octopus is having those random major twitches while moving. 
- I tried different USB port for power to the Delanclip and for the PS3 camera with the same result.
- Reinstalled Opentrack and tried the default profile. 
- I don't see any problem with the point extraction. 
- I can't feel any FPS drop using CL-Eye or Opentrack. I'll check if I can get some numbers instead of going by feel.

Since you're having the same problem as me and the same time, something feels off here. Maybe a Windows update? 

Let me know if you find anything, I'll do the same. 

Thanks!

freecodecampster, 2021-03-01:
I did get a similar jerky octopus with a PS3 Eye Camera and Aruco paper marker tracker setup. Turns out it was due to me messing around with camera settings. Switching to another profile with the camera settings listed below I was able to see much smoother movement. I'll provide my camera settings just in case they might work for you.

Camera settings
Exposure: 255
Gain: 30

Tracker settings
Diagonal FOV: 56
FPS: 60
Camera name: PS3 Eye opendriver
Resolution: 640x480

TC7270, 2021-03-07:
This just happened to me yesterday, I tried everything, but nothing was working. When I got up today, I decided to uninstall the opentrack and the PS3 camera completely. I then check DCS and saw the headtracker settings had no tracker. I then rebooted the PC. Then downloaded both software and re-installed it. Everything is now running great. Hope this helps......

MathijsG, 2021-04-17:
Had this with a client as well. He reinstalled or upgraded his Nvidia drivers and it worked again. You clearly see the FPS drop as well in opentrack.

khan1912, 2021-05-08:
On the PS3 camera, just move from Red to Blue or Blue to Red...not sure which one atm. That fixed it for me

mjtahir, 2021-07-03:
Had the same issue. FPS drops from 75 (640 x 480) to almost zero FPS. I've seen recommendations regarding changing camera settings, updating drivers, windows update, restarting PC, using different USB ports etc.

For me though, a simple Windows 10 update fixed the problem. Note I didn't rollback to a previous version, I updated to a newer version by "Check for updates" through Windows 10 settings.

sthalik, 2021-07-03:
What happens if you set opentrack to realtime priority?

iAmEMT, 2021-07-07:
After posting this issue, the problem seems to have solved itself, but now I'm having this issue once again after playing DCS only a few days ago. Camera FPS jumps from 60 to 23 to 4 while tracking. 

I've already tried solutions from this thread about changing camera settings, different USB ports, and I am on the most recent Nvidia drivers and opentrack 2.1.13. Setting opentrack to realtime priority also does not appear to solve the problem. The frame drops aren't noticeable when moving my head around in CL-Eye Test or in the tracker window in opentrack, so I'm not sure what would be causing this.

rellimmot, 2021-08-14:
I'm working with this problem now. Here's my working hypothesis: It appears to be an Nvidia drivers problem, some power saving mode it appears. Reason I say that is when I run something at 60FPS (https://www.youtube.com/watch?v=LXb3EKWsInQ) I can turn the PS3 Eye up to 90FPS just fine. When I close that video tab it goes back to stuttering badly.
When I make adjustments in NVIDIA Control Panel, when I hit apply the PS3 Eye preview video goes smooth but as soon as the settings are applied it goes back to heck. It appears in that transitionary period of applying settings, NVIDIA stops doing whatever optimization thing its doing that's causing PS3 Eye to crap out. I have had the problem happen in CL-Eye Test too so it may not be OpenTrack.
Some fix may lie in one of the settings in 'Manage 3D Settings' but I haven't found it yet.

Delanclip, 2025-01-07:
If you're experiencing jitter or stutter issues with your camera, it's likely a system-related issue rather than a problem with the camera itself. Here's how to troubleshoot:

**Disconnect USB Devices**
Unplug all USB devices from your system, leaving only the essentials connected: your keyboard, mouse, and camera.
Launch your game or software and check if the issue is resolved.

If it works smoothly, you might be experiencing USB bandwidth or power overload.

**Avoid Using USB Hubs**
Connect the camera directly to a USB port on your motherboard rather than through a USB hub. Hubs can often cause insufficient power delivery or bandwidth bottlenecks, which can lead to performance issues.

**Consider a PCIe USB Expansion Card**
If you need additional USB ports, it's recommended to use a PCIe USB expansion card. These cards provide dedicated power and bandwidth for each USB port, ensuring stable performance. Here's an example of a reliable card: [PCIe USB Expansion Card Example](https://www.amazon.co.uk/gp/product/B08PF8XR73)

Feel free to share your results or ask further questions! This process has helped many resolve similar issues.

closed 2025-02-25 by iAmEMT