opentrack/opentrack #1235: Update build documentation

gavazquez, 2021-02-14:
Hi @sthalik 

I would like to do a "mixed" tracker that combines a UDP tracker with an arduino and a BNO055 (a highly accurate pitch roll yaw sensor) and a camera tracker with 1 led for the X,Y,Z axis.

The thing is that the documentation for building under VS (https://github.com/opentrack/opentrack/wiki/Building-under-MS-Visual-C---2017-and-later) it's a bit outdated and looking at this issue: https://github.com/opentrack/opentrack/issues/1228 it also seems to be also a bit complex.

Would it be possible to update the wiki page and describe all the compilations steps? Or make a .sln file or something similar to compile on visual studio?

Thanks a lot!

sthalik, 2021-02-15:
> Would it be possible to update the wiki page and describe all the compilations steps? Or make a .sln file or something similar to compile on visual studio?

No, you have to generate .sln for Visual Studio. Or better yet, use native cmake support as any single build system change will overwrite all .sln and .vcxproj files.

> looking at this issue: #1228 it also seems to be also a bit complex.

The project has lots of dependencies and nothing can fix that.

Slion, 2021-02-15:
Took me a while to find the documentation I put up on wiki a while ago, hopefully that can help you some:
https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows

gavazquez, 2021-02-15:
That's very helpful guys. I haven't touched C++ since quite a lot of time (I mainly work with C#) but I'll try with VS2019 and if I discover something different during the build process I'll let you know.

GO63-samara, 2021-06-03:
I shared my experience with building Opentrack from source here:
https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-build-from-source
The description is in Russian, but there is also a videos with English titres.
Maybe this will help.
Good luck!