opentrack/opentrack #1236: Wine output protocol; version selection?

bradley-r, 2021-02-16:
I was planning to use Linuxtrack when my new tracker arrived, and while it does detect the tracker, no amount of axis tweaking, inversions and physical camera positioning can solve the (unforeseen) erratic and incorrect movements. Opentrack (installed via AUR) detects it perfectly, and also sends this data to my game (DCS in this case.) 

I run my games via Lutris, and thus much of their configuration is done through it, rather than directly editing every game's Wine prefix. Lutris allows the choice of various different Wine versions for each game, and this is where my issue arises. I run DCS with Wine version `lutris-6.0-x86_64`, while my system Wine is at 6.2. The Wine output in Opentrack seems only capable of using the system version, which if used in this case, prevents the game from loading thanks to a wineserver version mismatch.

I _can_ run DCS with my system's Wine install, but I prefer to keep any necessary tweaks and edits isolated to a per-game basis. Is there any way to select a Wine version for the output protocol to use? A search through previous issues didn't turn anything up, so I could understand if this has never been thought of before, but I at least would find an option like this very useful. 

Thanks for your time.

bradley-r, 2021-02-16:
Some more searching brought [this](https://github.com/opentrack/opentrack/pull/996) to my attention; the introduction of the Proton version options.  Effectively, I'm looking for what's exactly described [here](https://github.com/opentrack/opentrack/pull/994); a selection box just like that for Proton, but to select different Wine versions by pointing directly to their install path, in Lutris' case `/home/[USER]/.local/share/lutris/runners/wine/`.

bradley-r, 2021-03-03:
Going to close this for now, as I've managed to get Linuxtrack to work similarly well with my new tracker (by tinkering with some V4L2 settings), with it's implementation allowing multiple different Wine versions to be used across multiple games.

closed 2021-03-03 by bradley-r

sthalik, 2021-03-03:
You can't select different Wine versions with opentrack due to ABI breakage in libwine.

bradley-r, 2021-03-03:
> You can't select different Wine versions with opentrack due to ABI breakage in libwine.

I thought as much, but was worth the ask. Thanks for the clarification!