opentrack/opentrack #1239: Output Stuttering In Filtering Not Solved By Update (#1218)

mlb406, 2021-03-07:
My problem is exactly the same as described in Issue #1218. The raw received data is fine, but the game data sent from Opentrack to the game is extremely buggy and stuttery, jumping around and skipping to different points.  [https://youtu.be/CDcimn5LgPk](I made an unlisted YouTube video) that shows the problem, however the issue is identical to that in issue #1218. I am also using the smoothtrack app on my phone which sends the facetracking data to opentrack. 

Unlike issue #1218, my problem persists despite me being on Windows 10 version 2004. The original issue claims that Windows version 1909 fixed their problem, but I am on a later version of Windows, with no new updates available. There have been no major changes to my system between Opentrack working fine two days ago, and this problem arising this morning.

sthalik, 2021-03-07:
Disable game mode, increase process priority.

sthalik, 2021-03-14:
Also restrict a game to less CPU cores than you have available.

QuantumFractal, 2021-03-15:
Hey @mlb406 I had this same issue with my setup (Smoothtrack on Iphone11 / iOS 14 and Opentrack 2.3.12). I'm on Windows 10.0.19042 Build 19042 still. I noticed that my desktop's power settings were on "balanced" after setting it to "high power", then installing the latest Win10 updates Opentrack is running smooth again. Hope this helps!

jellopudding, 2021-03-30:
I recently had this issue; I was using AItrack and Opentrack 2.3.12 and found that Accela was particularly finicky (other filters were had similar issues but playing with the settings you could compensate for the majority of stutter, spikes, and other erratic behavior). I found that other applications, like firefox, discord, etc. were also having issues and seemed rather unresponsive before I updated/rebooted. If an updatre/reboot wont fix this issue for you, it might be that you have some other application or malware hogging resources.