opentrack/opentrack #1241: [Feature request] Add max output for yaw, roll, x, y and z

bababoombatman, 2021-03-12:
In the mapping menu we can lower the max output for pitch to 90°, but this option is not present in the yaw, roll, x, y and z menu. Would it be possible to add them? It would help with precision mapping when higher values are not needed.

Unrelated, but i would also like to ask if it's possible to add even lower values to pitch max output setting. For example 45°.

Thanks in advance

sthalik, 2021-03-14:
No, almost everyone needs full motion range. There's no need to clutter the UI.

diVineProportion, 2021-03-26:
@bababoombatman you know you can do this right? 
[image]