opentrack/opentrack #1244: Webcam MJPEG Mode

SqSqCivilprotection, 2021-03-17:
Hello,

I am trying to modify a highspeed webcam to use as an IR tracker. The webcam is capable of recording at 100-120 fps at 640 x 480 resolution in the MJPEG format, but only 30 fps in the YUY2 format. Opentrack only seems to be able to read my camera output in the YUY2 format. Is there any way that Opentrack (like AMCAP) can force my webcam into MJPEG mode while running? Any help would be appreciated.
Cheers!

Resources:
[image]
[image]
[image]

Webcam Link: https://www.amazon.com/gp/product/B00KA7WSSU?ref=ppx_pt2_dt_b_prod_image
[image]

sthalik, 2021-03-18:
Look for opencv's `cap_dshow.cpp`, based on venerable videoInput.

SqSqCivilprotection, 2021-03-18:
Thank you for the quick response!

This is what I found: https://github.com/opencv/opencv/blob/master/modules/videoio/src/cap_dshow.cpp

However, I don't know what to do with this (I am new to github). I give it more of a look tomorrow with fresh eyes, and maybe ask a couple other people.

Thank you again for your help.

sthalik, 2021-03-18:
You should be able to open camera settings through PT tracker's dialog. It's on the first tab.

SqSqCivilprotection, 2021-03-18:
[image]
 You are talking about these options yes? I have looked several times, and I cannot find an option to change the camera from YUY2 to MJPEG.

SqSqCivilprotection, 2021-03-20:
I have spent a couple of days trying various "fixes" and work-arounds, and haven't found any success. @sthalik do you know of anyone who has used a high FPS camera that isn't the PS3 eye cam?

sthalik, 2021-03-21:
This is interesting:

```c++
cap.set(cv::CAP_PROP_FOURCC, cv::VideoWriter::fourcc('M', 'J', 'P', 'G'));
```

We could try switching to MJPEG for dshow when FPS is higher than 30 and the camera isn't a PS3 Eye. I'm going to look into it once I go back into working on opentrack.

SqSqCivilprotection, 2021-03-22:
I think it would be a very good thing to explore MJPG in Opentrack. I'm trying to put together an updated DIY video about how to build a head-tracker. The PS3 eye camera is getting harder and harder to find, and is priced as high as $40 sometimes. I don't know for how much longer I can recommend it. I figure it is a good time to start exploring other cameras for tracking.

Anyway, I wanted to thank you for all the time and thought you have put into this issue. I'm satisfied to close this thread, and I look forward to future Opentrack updates.

Thank you again, Cheers!

N7-Hash, 2021-03-31:
Yes, please let that happen. I'm also excited about the possibility of future MJPEG support in OpenTrack. I bought my IR camera (**MagicIR**) from eBay about a year ago, and the seller told me it could capture at 640/480 120fps, but he didn't tell me it wasn't compatible with OpenTrack. I'd love to be able to use my IR camera fully. Any support for MJPEG would be greatly appreciated.

Sephirotic, 2021-07-19:
I have the exact same camera and opened a similar ticket. From what I could figure, this is not an issue of being MJPEG or YUY2, but about the video camera driver always initializing in 1080p30fps, otherwise you wouldn't get 30fps in 1280x720p fps mode but only 6. When you change the capture pin settings, after you close and re-opens the camera in many software, it goes back to the default 1080p30fps resolution. if you could enforce a "camera pin" change when initializing the camera driver when clicking "start", than this issue would be fixed. From what I could tell, the manual resolution and FPS you set up with Open Tracker has no effect on the camera mode, it just crops the image from sensor. 

https://github.com/opentrack/opentrack/issues/1298#issuecomment-882117339
This is a very nice camera for open track, it has a very easily removeable IR filter, manual exposure, low input lag, good SNR 720p@60fps or even 480p@120hz image, and interchangeable lenses, not to mention being quite cheap under 50 dollars with a 2.8-12mm lens. 

I tried using OBS Virtual Cam and I managed to feed 60fps input into Open track, but ufortunately this daisy-chinning created noticeable larger input lag.

Sephirotic, 2021-07-19:
> I think it would be a very good thing to explore MJPG in Opentrack. I'm trying to put together an updated DIY video about how to build a head-tracker. The PS3 eye camera is getting harder and harder [...]

This is not a issue regarding not being able to choose MJPEG mode, but about Point Tracker not correctly triggering the right Capture Pin modes and ignoring the FPS field. 

I've found a workaround. 

Whenever I put 640x480 @ 60 or 120 fps (which are actual modes in the camera pin), the "status" info when initializing the point tracker was always 30 fps. It didn't respect the FPS I had chosen. The same Happened for all other actual resolutions: 1280x720@60fps, or  800x600@60 fps. I was getting 10 for 720p and 20 fps for 800x600 (which indicated it actually was working in YUY2 mode like the other user suspected). However, when I changed the resolution to an uneven number close to the actual resolution I wanted, for example, 1200x720, @ 60fps, then the camera pin switched to the correct mode: 1280x720p@60fps (and MJPEG). 
I tried 640x480 @120hz, and I got around 75~81fps (oscillating around this. It seems Point Tracker can't process 120fps even with the high single core boost of a 5900x).  The result is awesome in DCS, the quality of the camera is much superior than the PS3eye and having better Dynamic Range and sensitivity also meant I could make the exposure darker so that it could isolate the IR leds even with the window open in a sunny day (with sun rays being blocked by a curtain, of course). 
I have made two videos comparing the PS3 eye camera with the ELP OV2710 Hd Usb camera with CS Mount here:

https://youtu.be/ELABP2SgUyw
https://youtu.be/RNwth7gtsc4

The second video tests input lag with a 240fps slow motion capture from my smartphone. Actual lag is about 13 frames (or 50ms) for the ELP camera in both 640x480@120hz mode or 1280x720@60, and 17 frames for the PS3 eye camera (70ms, 20 more than the ELP HD camera.) 

Camera was 60 bucks from aliexpress, considerably more expensive than tha PS3eye camera, but worth it IMO. 
If the developers of Point Tracker could fix their capture-pin mode selections this would be great.

Popertoeper, 2021-09-07:
Hi guys,

are you able to get more than ~72FPS with the ELP camera now that the MJPEG Compression feature was added with the release 2021.1.0 #1318 ?

I am not getting more than 72FPS... At least the workaround is not needed anymore:)

IrishDeviant, 2023-08-17:
Has there been any update to this? I'm using an arducam that can run 120fps, but no matter what I set the fps to in the settings, the camera status still reads out 75fps. Is this a hard limit of Opentrack? If so, why is it capped?

sthalik, 2023-08-17:
There's no hard limit anywhere. I'm using 640x480@60Hz on my IR webcam in MJPEG mode.

IrishDeviant, 2023-08-17:
> There's no hard limit anywhere. I'm using 640x480@60Hz on my IR webcam in MJPEG mode.

I tinkered with it a bit, and by manually entering the resolution like other have mentioned, my cam will run at 105fps, but it's not getting up to 120fps, which is what the cam is rated at. Even though I manually typed 120 in the fps field. Is there something keeping it from reaching 120?

sthalik, 2023-08-17:
Single-core CPU usage is the limit. With the PT tracker you may want to disable auto thresholding to save some time.

IrishDeviant, 2023-08-17:
Roger that. 105-110fps is probably sufficient, I doubt another 10fps would be noticable. I was mainly just curious if I was missing something with the setup.

sthalik, 2023-08-17:
You're going to get more jitter if you don't cap the framerate below max CPU usage.

IrishDeviant, 2023-08-17:
I'm not seeing any jitter but I haven't used it with a game running so I'll cap it at 75 and call it good.

closed 2023-11-05 by sthalik