opentrack/opentrack #1246: opentrack-wrapper-wine crashes when used with wine 6.x

XenoPL, 2021-03-19:
As per title opentrack-wrapper-wine.exe.so process crashes just after launch. If started from console with
 env WINEPREFIX=<my_wine_prefix_path> wine /usr/lib64/opentrack/opentrack-wrapper-wine.exe.so
it crashes spitting out message:
 00fc:err:seh:segv_handler_early Got unexpected trap 14 during process initialization

Once wine is downgraded to 5.x (here tested with 5.18) wrapper process starts just fine and stays on until headtracking is stopped. 
Tested with opentrack 2.3.12 and 2.3.13, wine 6.3.

Steps to reproduce:
 - start opentrack
 - select pointtracker as input, accela as filter wine as output
 - configure wine output plugin and point it to wine prefix where app that uses headtracking is installed
 - start headtracking
 - start app/game from 

Expected result:
 - opentrack-wrapper-wine.exe.so stays in the list of active processes
 - headtracking works in game/app
 
Actual results
 - opentrack-wrapper-wine.exe.so process appears for a brief moment then instantly closes
 - headtracking works in opentrack but not in a game.

XenoPL, 2021-03-30:
BUG also reported on WINE bugzila: https://bugs.winehq.org/show_bug.cgi?id=50892

XenoPL, 2021-04-01:
Small update, suspicious releases  narrowed down to wine 6.3. Opentrack under Wine 6.2 installed from Fedora rpms works fine.

XenoPL, 2021-04-10:
Last know working version of wine is 6.2, grab it from koji:
https://koji.fedoraproject.org/koji/packageinfo?packageID=4106

XenoPL, 2021-04-14:
I've tried wine 6.6 (available in updates-testing), and it seems issue is fixed, wrapper process stays on and headtracking is working. However in BMS 4.35u1 ,mouse doesn't work in 3D world (no cursor, unable to click anything). But it doesn't seem to be related to opentrack in any way.

closed 2021-04-14 by XenoPL

Leaguesman, 2021-06-19:
I'm getting this error now with Wine 6.10, though there's no longer any error message presented, the process exits a few moments after starting tracking. I previously had it working with Wine 5.12 before I did a system upgrade, but downgrading it didn't help, and neither did rebuilding opentrack against either Wine version.