opentrack/opentrack #1247: Proposal for image based tracker using custom AI / neural nets

DaWelter, 2021-03-23:
Hi

I've been working on neural-network based head-pose detection from RGB/webcam video which I would like to contribute to the opentrack project. It is still work in progress. 

Here is a link: https://github.com/DaWelter/opentrack/releases/tag/tracker-neuralnet-0.0
Release notes include link to demo video.

My work is very much inspired by AITrack https://github.com/AIRLegend/aitrack and https://github.com/cleardusk/3DDFA_V2.

There are a few points I'd like to mention. In no particular order.

* I want to release the training code of the networks eventually. Thus we would have a full open source baseline for RGB-only head-tracking. Right now it is a mess though. And I won't ask for a merge until full disclosure.

* Should run on one core of an i7 2600k cpu at 30 fps.

* I did not know about AITrack, nor ViewTracker, nor SmoothTrack. When I learned about facial keypoint detection I figured I knew the perfect problem for my solution ... ;-D

* This competition looks very strong. My tracker might not be as good but there is value in having an integrated solution.

* Depends on the ONNXRuntime library. Fortunately it can be built from sources relatively easily:
```
$ build.bat --config RelWithDebInfo --x86 --build_dir .\buildx86\ \
 --enable_msvc_static_runtime --build_shared_lib --skip_tests \
 --cmake_generator "Visual Studio 15 2017"
```
... in order to get a x86 dll. Their build script pulls all deps automatically, and the final dll depends only on standard system stuff. :-) I built the v1.6.0 release. Furthermore, https://github.com/microsoft/onnxruntime/releases/tag/v1.5.2 still contains prebuilt x86 libs which should be compatible, too.

* Depends on Eigen. Still. Will get rid of this probably.

* For the computer vision practitioners: Under the hood is a Mobilnet V1 which outputs 4 quaternion components, 2d coordinates of a location near the nose, and the diameter of the head in terms of pixels. Strangely, I found Mobilnet V1 to compare favorably against more modern architectures.

* Doesn't actually perform facial keypoint detection as AITrack does. Well, not in its primary function. My idea was to output the head rotation directly rather than estimating the pose through cv::solvePnP based on keypoints. However, the network also outputs parameters for a deformable model from which keypoints can be taken. These are used in training, trying to fit keypoints to the training data.

* I trained the network on a combination of various datasets: Biwi, AFLW2k, YouTube Faces With Facial Keypoints, and 300W-LP. If I understand it right, at least some of them forbid commercial use.

* The tracker might be racist and won't work equally well for everyone due to bias in the training data.

* 3DDFA works better tbh. It has a better training algo and better (synthetic) data. I will try to improve my network though. 

* Here is a script to send the pose from 3DDFA to opentrack in the same way as AITrack does. https://github.com/DaWelter/3DDFA_V2/blob/master/demo_opentrack.py I used that as reference to get a feel for what's state of the art. It's Python though and you need pytorch and so on to run it.

What do you think? Can you imagine that tracker to get integrated?

Cheers
Michael

GO63-samara, 2021-03-24:
I checked it, the **Neuralnet tracker** really works. CPU load 9%. The **Neuralnet tracker** needs to be including in Opentrack.

DaWelter, 2021-03-25:
Thanks for checking and the favorable review. :-)
I was on Skype with Stanislaw, he will try it as well.
A few years ago I worked on the point tracker, hence i know him, and the project.

DaWelter, 2021-04-22:
Hi!
Finally done cleaning up the train code a bit. It is now published here https://github.com/DaWelter/neuralnet-tracker-traincode

Many things changed so I had to regenerate the networks which are included in the new release here https://github.com/DaWelter/opentrack/releases/tag/tracker-neuralnet-0.1

I play tested it and found it maybe even a little less noisy than before. Provided the illumination is good enough so that it can see something.

Before a PR there are still a few things I need to try ...
* Remove dependency on Eigen
* Automatic brightness amplification
* Automatic head center calibration

Issue I noticed during my tests: The network thinks the head grows bigger when you open your mouth. This is bad because the position computation depends on it. The network is generally sensitive to grimacing :-< Something I need to work on, too.

DaWelter, 2021-04-27:
I wrote the Kalman filter. It was a mistake. The accela filter works much better. It should be removed tbh :-|

It was not a big deal to change. Had to make several other changes as well which worked well together.

https://github.com/DaWelter/opentrack/releases/tag/tracker-neuralnet-0.2

I think I'm ready now for a PR.

DaWelter, 2021-04-30:
To be continued in a proper PR

closed 2021-04-30 by DaWelter