opentrack/opentrack #1249: GTA V - Mouse Emulator not working.

GallopingGepard, 2021-03-25:
Hey, so I wanted to combine driving in first person with my headtracking. I've been using the program for several years and never run into any issue until now. I'm on 2.3.13. 

GTA V uses the mouse to look around whilst in vehicles. I figured if I set OpenIV up to emulate the mouse it would work but I can't figure it out.

My input is: Joystick Input (which takes the input from EdTracker, works fine as always)
My output is: mouse emulation
My filter is: Accela

Everything appears working as intended but the output doesn't seem to have any effect in GTA V. In the mouse settings of the game I selected Windows and Input but neither changed anything.

sthalik, 2021-03-25:
Switch mouse mode in GTA 5 options.

GallopingGepard, 2021-03-25:
> Switch mouse mode in GTA 5 options.

Thanks for quick reply. Could you clarify? I have tried all three modes: DirectInput, Windows,  and Raw Input.. without success.

GallopingGepard, 2021-03-25:
[image]

sthalik, 2021-03-25:
DirectInput, then switch opentrack's mode to legacy.

GallopingGepard, 2021-03-25:
> DirectInput, then switch opentrack's mode to legacy.
> 
> Either run both opentrack and GTA as admin or neither.

Thanks! Changing the mouse emulation to legacy and sorting the X and Y axis to yaw and pitch works. 

Unfortunately not very well. It's pretty much unplayable.

MathijsG, 2021-04-11:
Well GTA isn't really meant for headtracking, so it would mostly just use the camera axes of the came.

I used it in-car in GTA V first person mode and it worked okay.