opentrack/opentrack #1255: Test Tracker, add option to send positional data

iwubcode, 2021-04-10:
I am developing something called a "DSU" which sends opentrack data to various game emulators.

To test this capability, I've used the test tracker to generate data.  Unfortunately, the test tracker only sends rotational data.  It'd be nice if there was an option to send positional data too  (or maybe always send it, since you can disable sending certain data in OpenTrack).

sthalik, 2021-04-10:
Fill in elements 0, 1, 2 on the data array.

GO63-samara, 2021-04-10:
Hey!
I usually use Dummytrack via UDP over the network to test Opentrack:
https://sites.google.com/site/diyheadtracking/home/opentrack/dummytrack-6dof-tracker-simulator-and-6dof-pose-indicator
Any 6DOF coordinates can be set using the sliders.
Maybe this will help.

iwubcode, 2021-04-10:
@sthalik - thanks.  I know that but just was hoping for something dynamic (like the test tracker).  The tool @GO63-samara mentioned will work.

@GO63-samara - thanks, works very well.  Is the source code available anywhere?

GO63-samara, 2021-04-11:
To make "game data" and "raw data" identical, you need:

- In "Options" set "Center at startup" to OFF.
- In the "Mapping properties" for the X, Y, Z axes, select Max input 100cm and set the point input 75 = output 75.
- In the filter, you need to set Dead zone = 0 or completely disable the filter.

If these three points are met, then "game data" and "raw data" must be identical.

GO63-samara, 2021-04-11:
I forgot the most important thing - you need to set "Relative translations" to "Disabled".