opentrack/opentrack #1256: First some .dll issues and now, can't launch

Lord-Hinater, 2021-04-13:
Hi, 
I had some .dll issues with opentrack. MSVCP140.dll and VCRUNTIME140.dll was missing. I tried to uninstall et re-install VS C++ but nothing. I tried to download manually thes .dll files, now the error is not about these dll files missing. But it shows me an error : 0xc000007b. I don't know how to solve it.
I tried with older version of opentrack but when i've dcs launched, opentrack cannot open. 
For information, i rebooted my computer yesterday and it is since this reboot i've these issues.
[image]
[image]
 and now :
[image]

GO63-samara, 2021-04-14:
Try the dll from this archive:
https://sites.google.com/site/diyheadtracking/home/opentrack/opentrack-download/OT-DLL.rar

This archive contains five variants of the often missing msvcp140.dll and vcruntime140.dll libraries.
I have dlls from the "checked" folder running on Windows7 x64 and on Windows10 x64.
These dlls need to be copied to the Opentrack folder.

Good luck!

Lord-Hinater, 2021-04-15:
ok cool thanks to fix my opentrack but now, it's my dcs which have the 0xc000007b error.... I'm desperate

GO63-samara, 2021-04-15:
> I tried to download manually thes .dll files, now the error is not about these dll files missing. But it shows me an error : 0xc000007b. I don't know how to solve it.

So the problem appeared after you manually wrote down the .dll files? Where did you write it down? Maybe now they need to be removed?

Lord-Hinater, 2021-04-26:
I just discover today i have the same issue with an other program... I'm waiting my High speed connection to re boot my computer and re try all of what we spoke about. Thanks for your helping !